malmeloo/FindMy.py #23: Apple server does not honer time window

aircable, 2024-04-08:
I'm trying to get data only for one day by setting startdate and enddate for the search.
This was the query. You see it's 24 hours.
`{'search': [{'startDate': 1712449083967, 'endDate': 1712535483967, 'ids':...`
But the Apple server always returns a week of data.
Is the format wrong?
Your code in accounts.py changed:
```
def fetch_last_reports(
        self,
        keys: Sequence[KeyPair],
        hours: int = 1 * 24,
    ) -> MaybeCoro[dict[KeyPair, list[LocationReport]]]:
```

malmeloo, 2024-04-08:
This is a known issue, see https://github.com/biemster/FindMy/issues/7 for more context.

I've briefly tried looking into this myself as well, but haven't figured out yet why it doesn't work. I think it might be an issue on Apple's side.

You can always manually filter the results if you wish, there's not a huge downside to fetching too many reports. I'd rather not do this in the library because throwing out a bunch of reports would be rather wasteful.

aircable, 2024-04-08:
Thank you for the reference, I didn't find it. 
I sort it myself and it's not a big deal. 
You may close the issue. When @biemster finds the issue we can fix it too.

closed 2024-04-08 by malmeloo