malmeloo/FindMy.py #24: Is OfflineFindingScanner supported on Windows?

darthnithin, 2024-04-13:
Is OfflineFindingScanner meant to work on windows?
If so, it doesn't and also it doesn't give a graceful error message
[image]

darthnithin, 2024-04-13:
[image]
makes me think it's not, but it would be great if there was a more relevant error message. Also "platform agnostic" is in the readme which is a little confusing?

malmeloo, 2024-04-13:
Thanks for reporting this! It should indeed work on Windows, but I haven't actually been able to test that myself. It looks like the library behaves a little differently on Windows compared to Linux/Mac OS, which is causing the issue.

Unfortunately I don't have a lot of time to look into this right now, but I intend to pick development back up very soon and will fix this asap.

darthnithin, 2024-04-14:
[image]
macos returns this tuple
[image]
while windows returns a [BluetoothLEAdvertisementReceivedEventArgs](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementreceivedeventargs?view=winrt-22621)
[image]
[image]

malmeloo, 2024-04-23:
This should now be fixed on main and will be included in the next release which is scheduled for this week. Please let me know if the issue persists.

PR #25 (2024-04-23): scanner: Fix Windows compatibility
Found `OfflineFindingDevice`s on windows will now report no additional device data due to platform constraints.

Fixes #24
closed 2024-04-23 by malmeloo