malmeloo/FindMy.py #36: can't retrieve location reports from third party 'Airtag'

snomile, 2024-05-12:
I've test real_airtag.py with some third party 'Airtag compatible' AIYATO tag(link is [here](https://www.amazon.com/Bluetooth-Tracker-Locator-Anti-Lost-Positioning/dp/B0BLCKF54S?th=1)), and I can't retrieve any location reports.
The tag is compatible with Findmy network, I can locate the tag via FindMy app, all functions go well and very smooth, no difference compared to Airtag.

I also tested other devices such as AirPods Pro, MacBook Pro and iPad, they all work fine with real_airtag.py.

This is part of plist of AIYATO tag, please tell me what information should I provide more, thanks!
[image]

malmeloo, 2024-05-14:
Interesting... it *should* just work, as 3rd party tags are functionally compatible with the official ones, and this library does not make a distinction between them.

One thing to note though, when you check the FindMy app, was the reported position near one of your Apple devices? Devices will record at which location they last connected to the tag, and will show that location instead if is more recent than whatever the network reports. They will also not be "findable" by other people's devices as long as they are in your proximity.
This library only fetches locations reported by the actual FindMy network, so it will inherently generate fewer reports than what you will see on your devices.

snomile, 2024-05-15:
thanks for you reply.
I've tested two tags, one is near my iPad while another tag is several hundreds meters away, they all shown in FindMy app and both no luck on retrieving location reports.
then I unpair one tag from my iCloud account, pair with another iCloud account, decode the plist from new account, suddenly I can retrieve location reports!
I must say it's very strange, and hope these information is helpful.

malmeloo, 2024-05-16:
That is very strange indeed, maybe the FindMy app on mac still had the wrong information cached?

As I understand it everything works now, correct? If so I will close this issue.

snomile, 2024-05-16:
tried to decode plist from different Mac, they're all the same, so I don't think the plist is staled.
another wired thing here, after changing the "hours" from 7*24 to 1, I suspect the server will give me reports of last 24 hours, but instead there're only part of it.
If I restore the hours from 1 to 7*24, I can get the missing reports which almost one report per minute.
Does that suggest there's something wrong with the public key generation? 

```
    def fetch_last_reports(
        self,
        keys: KeyPair | Sequence[KeyPair] | RollingKeyPairSource,
        hours: int = 1  
    ) -> list[LocationReport] | dict[KeyPair, list[LocationReport]]:
        """See `AsyncAppleAccount.fetch_last_reports`."""
        coro = self._asyncacc.fetch_last_reports(keys, hours)
        return self._evt_loop.run_until_complete(coro)
```

```
Location reports:
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:01:58+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:03:42+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:07:27+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:09:01+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:13:46+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:17:31+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:22:26+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:26:01+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:26:55+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:30:19+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:32:43+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:36:37+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:44:49+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:46:06+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:48:04+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:49:59+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 04:54:42+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 05:09:13+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 05:25:55+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 05:30:24+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 05:40:18+08:00, lat=
 - KeyReport(key=/bOztVZTt4YZ3YS+3Rh0BbAP/F5KEMZgJm9NQnvF8tU=, timestamp=2024-05-15 05:46:22+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:03:22+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:04:08+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:06:36+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:07:18+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:07:20+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:08:34+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:08:55+08:00, lat=
 - KeyReport(key=VyToc4OkZK41y+1+2oZZcaY34KbR4VvCX8RrETH79cc=, timestamp=2024-05-16 04:09:09+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:06:28+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:06:48+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:07:22+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:07:24+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:08:30+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:09:30+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:10:38+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:11:46+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:24+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:24+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:30+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:34+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:50+08:00, lat=
 - KeyReport(key=qglI3NXRSDKlrUDn6XYPez6fVZMq2Grmp1TayqJ7I0k=, timestamp=2024-05-16 23:12:52+08:00, lat=

Process finished with exit code 0
```

malmeloo, 2024-05-16:
The `hours` attribute is known to be "broken"; Apple's server does not seem to honor it. I'd recommend to just filter out the reports you don't need yourself. Changing that attribute simply asks Apple to send more or less reports, but the key generation remains unchanged.

snomile, 2024-05-17:
OK，thanks.

closed 2024-05-17 by snomile