malmeloo/FindMy.py #37: Add MAP to HomeAssistant

wes1993, 2024-05-13:
Hello @malmeloo,
I have created a script that can export all the location then send a HTML file to HomeAssistant:
[image]

Here is the code, If you want try :-D
```
"""
Example showing how to fetch locations of an AirTag, or any other FindMy accessory.
"""
from __future__ import annotations

import logging
import sys
from pathlib import Path

from _login import get_account_sync

from findmy import FindMyAccessory
from findmy.reports import RemoteAnisetteProvider

######### CSV and MAP CREATION
import csv
from gmplot import gmplot
from math import radians, cos, sin, asin, sqrt
from dateutil import parser as dtparser

######### Send to HomeAssistant
import paramiko
from paramiko import SSHClient
from scp import SCPClient

########################################### Configuration ###########################################
# URL to (public or local) anisette server
ANISETTE_SERVER = "http://localhost:6969"

# Name of the Airtag, used also for naming CSV file and HTML map File
person = "Simba"

# Map Configuration
# Home LAT AND LON, See Google Maps or HA to find this Info
LAT = 41.80
LON = 12.62

#ZOOM level, where 0 is fully zoomed out. Defaults to 13.
ZOOM = 12

# GMAPS ApiKey https://cloud.google.com/maps-platform/ click Get Started. Choose Maps and follow the instructions. Set restrictions for the API Key (Click on Maps Javascript API > Credentials > Your api Key > API restrictions and select Maps Javascript API).
apikey = ""

# Create PLOT, Values are True or False
CreatePlot = True
# Color of the PolyLines, Can be hex (‘#00FFFF’), named (‘cyan’), or matplotlib-like (‘c’), see here https://github.com/gmplot/gmplot/blob/master/gmplot/color.py and ColorCode.pdf: 
PColor = 'cornflowerblue'
#Width of the polyline, in pixels. Defaults to 1.
PWidth = 1

# Marker COLOR Can be hex (‘#00FFFF’), named (‘cyan’), or matplotlib-like (‘c’) see here https://github.com/gmplot/gmplot/blob/master/gmplot/color.py and ColorCode.pdf: 
# All Markers
#AMColor = 'red'
AMColor = '#FFFFFF'
# Keep Disabled until Fix Found, If Enabled the Last Location not evident
AMColorDisabled = False
# Last Location Marker Color and Text Color see here https://github.com/gmplot/gmplot/blob/master/gmplot/color.py and ColorCode.pdf: 
#LMColor = 'chartreuse'
LMColor = '#00FF00'
MARKERLABEL = False
# Number of last marker, for print all marker set to False
NMARKER = False
# Dark Mode for the MAP, Values are True or False
DARKMODE = True
[... 276 lines cut ...]
    print("Send MAP To HomeAssistant Instance, Under PATH /config/www/findmy/"+person+"_gps.html" )
    ssh = SSHClient()
    #ssh.load_system_host_keys()
    ssh.set_missing_host_key_policy(paramiko.AutoAddPolicy())
    ssh.connect(hostname = HAHostName, port = HAPort, username = HAUsername, password = HAPassword)
    
    # SCPCLient takes a paramiko transport as its only argument
    scp = SCPClient(ssh.get_transport())
    
    scp.put("data/"+person+"_gps.html", "/config/www/findmy/"+person+"_gps.html")
#    scp.get('file_path_on_remote_machine', 'file_path_on_local_machine')
    scp.close()

if __name__ == "__main__":
    if len(sys.argv) < 2:
        print(f"Usage: {sys.argv[0]} <path to accessory plist>", file=sys.stderr)
        print(file=sys.stderr)
        print("The plist file should be dumped from MacOS's FindMy app.", file=sys.stderr)
        sys.exit(1)
    if FORCEMAP == True:
        createmap()
    else:
        openfiles()
        sys.exit(main(sys.argv[1]))
```

malmeloo, 2024-05-14:
That's so cool! I actually don't even have a functional tag right now, so I won't be able to try this myself (yet) 😅. If you put up the code somewhere I will link to it in the readme, there are probably more people who are also interested in this.

Informatic, 2024-05-20:
It should be fairly easy to create a "proper" custom device tracker component based on FindMy.py - that would also have an added benefit of device location becoming usable in automations and all. I may look into it sometime in the near future.

malmeloo, 2024-05-20:
I have the foundations of a custom integration working already with a limited login flow, haven't had much time to work on it again but I can publish my progress today or tomorrow.

guniv, 2024-05-23:
Just chiming in to say a custom integration for AirTags would be really great.

malmeloo, 2024-05-26:
Sorry for the delay, but I just published the integration to a separate repository: https://github.com/malmeloo/hass-FindMy

The code is from back in February, so I hope it still works with the latest version of Home Assistant. IIRC it's able to log into an account + 2FA but not much more than that, but it should be an OK foundation to start with.

malmeloo, 2024-07-11:
I'll close this issue now since it is unrelated to FindMy.py. However, if anyone wants to share similar projects to this one, let me know! I'm interested :-)

closed 2024-07-11 by malmeloo