malmeloo/FindMy.py #41: Not getting back the latest reports after a while

thisiscam, 2024-06-29:
Somehow after a while `fetch_latest_reports` are not getting back the latest reports any more and get stuck at some time point.

I've tried:
- Passing in different parameters to `fetch_latest_reports(hours=...)` none worked
- Re-decoding the plist file with the MacOS FindMy running and using the latest plist file, no luck

I have been polling with a 15 minutes interval --- did apple detect and put some kind of hold on my account?

malmeloo, 2024-06-30:
Can you enable debug logging (through the `logging` module and try again? I suspect this is caused by #27. I know of a fix but am extremely busy with study right now; I'm hoping to pick this project back up again by the end of next week.

thisiscam, 2024-06-30:
> Can you enable debug logging (through the `logging` module and try again? I suspect this is caused by #27. I know of a fix but am extremely busy with study right now; I'm hoping to pick this project back up again by the end of next week.

Thanks I have enabled debug logging and waiting to see any more information.
I am using [my branch](https://github.com/thisiscam/FindMy.py/tree/new_data_format_fix) --- it implements the fix user Informatic proposed in #27 . Seems like it does fix the payload parsing errors.

I did some more digging and I think maybe I have some misunderstanding about how the official apple airtag works? 
Is it possible for the airtag to not go through the crowdsourcing location service if my own device is nearby? Some times in the macOS FindMy app, I can see my airtag's location being updated recently but the I get no latest reports from the library.

malmeloo, 2024-07-11:
This should now be fixed with #47, which will be included in a release soon. If not, feel free to reopen.

closed 2024-07-11 by malmeloo