malmeloo/FindMy.py #48: Need to re-login every time

hkfuertes, 2024-07-11:
Hello, I'm building a simple flask api that converts tag location to openhaystack reports (so that I can tweak the app to use real airtags)... and so I have put toghether a docker-compose with findmy inside flask and anisette.
It works the first time I login... After that I get this message:
```bash
root@63ae2a4c766f:/app# python get_reports.py 
Traceback (most recent call last):
  File "/app/get_reports.py", line 4, in <module>
    reports =  get_reports()
               ^^^^^^^^^^^^^
  File "/app/app.py", line 50, in get_reports
    reports[name] = (get_report(name))
                     ^^^^^^^^^^^^^^^^
  File "/app/app.py", line 39, in get_report
    acc.fetch_last_reports(airtag),          
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/account.py", line 1077, in fetch_last_reports
    return self._evt_loop.run_until_complete(coro)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/account.py", line 680, in fetch_last_reports
    return await self.fetch_reports(keys, start, end)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/account.py", line 639, in fetch_reports
    return await self._reports.fetch_reports(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/reports.py", line 239, in fetch_reports
    reports.extend(await self._fetch_reports(date_from, date_to, chunk))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/reports.py", line 260, in _fetch_reports
    data = await self._account.fetch_raw_reports(start_date, end_date, ids)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/reports/account.py", line 594, in fetch_raw_reports
    resp = r.json()
           ^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/findmy/util/http.py", line 48, in json
    return json.loads(self.text())
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
ERROR:asyncio:Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x7b0291a2fc50>
ERROR:asyncio:Unclosed connector
connections: ['[(<aiohttp.client_proto.ResponseHandler object at 0x7b0291be3310>, 10243.288814196)]']
connector: <aiohttp.connector.TCPConnector object at 0x7b0291a2d5d0>
ERROR:asyncio:Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x7b0291a72f50>
ERROR:asyncio:Unclosed connector
connections: ['[(<aiohttp.client_proto.ResponseHandler object at 0x7b0291be2d60>, 10243.775111868)]']
connector: <aiohttp.connector.TCPConnector object at 0x7b0291a1bc90>
```

malmeloo, 2024-07-11:
Perfect timing! I just created #49 which introduces some improvements in this area. Can you update the library in ~15 minutes and try again?

hkfuertes, 2024-07-11:
Now it gives more info:
```
findmy.errors.UnauthorizedError: Unexpected login state after reauth: LoginState.REQUIRE_2FA. Please log in again.
```
Maybe there is away to use an app password to avoid 2FA...

malmeloo, 2024-07-11:
You mentioned in the other issue that you are using a self-hosted anisette server, how are you deploying it? It seems to me that the server is losing its state after a restart, which will cause it to hand out different information and log you out. If you are using anisette-v3-server, make sure you have the volume set up as instructed; if you don't and recreate the container, it will cause the issue you're running into now.

hkfuertes, 2024-07-11:
```yaml

services:
  server:
    build: .
    container_name: server
    ports:
      - 5000:5000
    command: /bin/sh -c "pip install -r requirements.txt && flask --app app.py --debug run --host 0.0.0.0"
    working_dir: /app
    volumes:
      - .:/app
      - ../PLISTS:/PLISTS
    depends_on:
      - anisette

  anisette:
    restart: unless-stopped
    container_name: anisette
    volumes:
      - anisette_data:/home/Alcoholic/.config/anisette-v3/lib/
    image: dadoum/anisette-v3-server

volumes:
  anisette_data:
```

malmeloo, 2024-07-11:
I can confirm your issue, but only when `docker compose down`-ing the stack before starting it again. It seems that the device provisioning data is not written to the volume, so if you destroy the container it is lost and the container will reprovision, leading to authentication errors.

The provisioning data is stored inside the  `/home/Alcoholic/.config/anisette-v3/` directory, so if you drop the `lib/` suffix from your volume specification, it should work.

hkfuertes, 2024-07-11:
Nice catch!!! I got it working after downing! Thank you a lot!

closed 2024-07-11 by hkfuertes

NickCrews, 2025-05-22:
Is there any way to make some sort of service account that a GitHub Actions bot can use to scrape an airtag's location every 15 minutes? I have a python script working on my macbook, but I am running into this issue. If I host my own anisette server, can that persist my login state?

The larger problem I am trying to solve is I want to set up a geofence to alert me if an airtag leaves some geographic area, if you have an alternative way to solve that, I'd love to hear. Thank you!

malmeloo, 2025-05-22:
Yes, you absolutely should be using your own Anisette server. Losing your login session when you're using public Anisette data is completely expected. If you use your own server, it should theoretically last for a very (very very) long time. I realize this might not be very clear from the information currently presented in the repository, but I'm working on better documentation 🙂 

If you want a DIY solution, you could probably throw the lat/long coordinates into the haversine formula to find out whether the airtag exceeds a certain distance from a given point. There is also https://github.com/malmeloo/hass-FindMy for if you have Home Assistant, but right now it only supports DIY FindMy tags. I'm not sure of any other plug-and-play solutions...

NickCrews, 2025-05-22:
Hurray, I just deployed https://github.com/Dadoum/anisette-v3-server to google cloud run on a personal google cloud account, and it worked. The persisted account.json seems to work now without redoing the 2FA.

I inferred the following steps from the given docker command on that repo: `docker run -d --restart always --name anisette-v3 -p 6969:6969 --volume anisette-v3_data:/home/Alcoholic/.config/anisette-v3/lib/ `

1. https://console.cloud.google.com/run and go `deploy > service`
1. choose `dadoum/anisette-v3-server` as the image
1. set `6969` as the port
1. create a volume using GCS. I created a bucket right there in the web ui, it was super easy. Mount it at `/home/Alcoholic/.config/anisette-v3/` (following your pointer above to drop the `.lib` suffix from the volume.)
1. Allow all traffic (don't require authentication etc)
1. Configure the service name, region, autoscaling, etc as desired.
1. It works!

NickCrews, 2025-05-22:
In case this is useful for others, I'm using duckdb for the spatial analysis, and configuring the geofence policies in a google sheet with a publicly accessible CSV url. Then duckdb can handle fetching that configuration since its so easy to fetch CSVs from urls with duckdb, as well as duckdb can do the spatial analysis for if coordinates violate the geofence policies. Then I can control the policies from the google sheets app on my phone or any web browser.

malmeloo, 2025-05-22:
Thanks for sharing! That's actually a really neat way to do it. Just FYI, I'm almost done with integrating #2, once it's released the entire Anisette server should no longer be a requirement since the library will be able to generate the data on its own. The state could then be entirely contained within the `account.json` instead of having to rely on external servers, which should simplify things a bit.

NickCrews, 2025-05-23:
That would be excellent and would really simplify things for me!