malmeloo/FindMy.py #59: coroutine 'AppleAccount.close' was never awaited

hkfuertes, 2024-08-02:
Hello, don't know if this is a bug... but sometimes I get this issue:

```bash
/usr/local/lib/python3.11/asyncio/base_events.py:679: RuntimeWarning: coroutine 'AppleAccount.close' was never awaited
  self._ready.clear()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
ERROR:asyncio:Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x7845015c8450>
ERROR:asyncio:Unclosed connector
connections: ['[(<aiohttp.client_proto.ResponseHandler object at 0x78450272cec0>, 2831.831099158)]']
connector: <aiohttp.connector.TCPConnector object at 0x7845015f3f50>
ERROR:asyncio:Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x7845015cf090>
ERROR:asyncio:Unclosed connector
connections: ['[(<aiohttp.client_proto.ResponseHandler object at 0x7845015b81a0>, 2832.221270141)]']
connector: <aiohttp.connector.TCPConnector object at 0x7845015f3a50>
```

Thank you for your work, I'm and android user, and through this I was able to pin point my tags in a Home Assistant map :)

malmeloo, 2024-08-02:
Can confirm, I run into this issue too sometimes. The sync wrapper attempts to clean up when an account object is gc'd, but for some reason the coroutine is never actually awaited, resulting in that warning. It's harmless, but quite annoying.

I haven't really found a solution to this yet, but thanks for filing an official bug report :-)

malmeloo, 2024-08-02:
Would you be willing to try #60 to see if it fixes the issue? i.e.:

```bash
pip install git+https://github.com/malmeloo/FindMy.py@fix/coro-never-awaited
```

It's been much more reliable for me, I hope it's a universal solution.

hkfuertes, 2024-08-02:
It seems to be working... I stressed the program... but no error have been shown...

malmeloo, 2024-08-04:
Nice to hear! I'll go ahead and merge the change then, new release should be out in a bit.

PR #60 (2024-08-04): fix: Unawaited coroutine errors on `Closable`
Should fix `Coroutine <xxx> was never awaited` errors and warnings on exit.

Closes #59
closed 2024-08-04 by malmeloo