malmeloo/FindMy.py #64: Exception on Python 3.9

siebert, 2024-08-24:
The pyproject.toml file indicates that FindMy.py should work with Python 3.9 and newer, but on Debian 11 (bullseye), which uses Python 3.9.2, I get this exception:

```
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import findmy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/findmy/__init__.py", line 2, in <module>
    from . import errors, keys, reports, scanner
  File "/usr/local/lib/python3.9/dist-packages/findmy/reports/__init__.py", line 2, in <module>
    from .account import AppleAccount, AsyncAppleAccount
  File "/usr/local/lib/python3.9/dist-packages/findmy/reports/account.py", line 14, in <module>
    from typing import (
ImportError: cannot import name 'Concatenate' from 'typing' (/usr/lib/python3.9/typing.py)
```

malmeloo, 2024-09-02:
Oh oops, I think on python 3.9 it needs to be imported from `typing_extensions`. I'll fix it and get some very basic CI tests going to prevent this in the future.

malmeloo, 2024-09-02:
Should be fixed in #71, which will be included in the next release.

closed 2024-09-02 by malmeloo