malmeloo/FindMy.py #72: Help?

laopaoer-wallet, 2024-09-03:
"I referred to your code and have some questions.

How do I generate the parameters X-Apple-I-MD, X-Apple-I-MD-M, and X-Apple-I-MD-RINFO?"

malmeloo, 2024-09-03:
`X-Apple-I-MD`:
base64 string encoding 28 bytes, 'one-time password', not sure how to generate this.

`X-Apple-I-MD-M`:
base64 string encoding 60 bytes, 'machine id', not sure how this is generated either.

`X-Apple-I-MD-RINFO`:
An integer representing 'router info'. From the [source](https://github.com/malmeloo/FindMy.py/blob/693c7c46570843b4cff37ea8277b9a9486d31c63/findmy/reports/anisette.py#L63C9-L67C60):

```
A number, either 17106176 or 50660608.

It doesn't seem to matter which one we use.
- 17106176 is used by Sideloadly and Provision (android) based servers.
- 50660608 is used by Windows iCloud based servers.
```

FindMy.py does not generate these values itself, but uses an external Anisette server to retrieve them. I'm still looking into using local options for this, by utilizing e.g. pyprovision.

malmeloo, 2024-09-11:
Closing due to inactivity, feel free to reopen if your question has not been answered.

closed 2024-09-11 by malmeloo