malmeloo/FindMy.py #88: Request "ring" the device

phamleduy04, 2024-12-02:
Like the title said is there any way to send ring/play sound request using this project?

malmeloo, 2024-12-02:
This is not currently possible, but it is something that I'd be willing to implement. That said, there are some network-related improvements which are higher priority at the moment, so it might take a little while before this is implemented.

I'll keep this issue open as a feature request!

parawanderer, 2025-03-19:
I'd like to second the "Play Sound" feature. 

Also not really a huge ask from me, but I observed that even my Android phone has some built-in Google Playstore thing that lets it "Play Sound" for "[Detected Unknown Airtags](https://www.idropnews.com/news/your-android-phone-will-soon-be-able-to-alert-you-to-an-unknown-airtag/197838/)", so it seems pretty intentionally easy to trigger this behaviour from them.

malmeloo, 2025-03-20:
Yes, there is a public BLE endpoint to request the device to make a sound. It's not very difficult to implement, but I'm still thinking of how to design the API around it in a 'neat' way. 'Authorized' BLE interactions are also on the roadmap, so this feature will probably drop around the same time as that.

saechsischercoder, 2025-06-26:
please, please, please implement this 🙏

oischinger, 2025-07-16:
Do I understand this right that this is "just" a BLE message and not using Apple's findmy Network at all?
If so: is there any Security protocol on thia BLE message?

malmeloo, 2025-07-16:
Yes, it's just a BLE message. There's not really any security involved, because other people should be able to ring the device as well. This should be relatively easy to implement, so although there are a few other issues with higher priority, I'm planning on tackling this one soon ish.

JealousEagle, 2025-07-19:
Hello , yes it would be a very nice feature ,thanks for sharing your project !!

malmeloo, 2025-10-15:
A rough sketch of my current TODO list:
1. AirTag support (+ other fixes) in hass-FindMy
2. Local AirTag communication (so this issue + maybe some more advanced features)

As always, no ETA.

> Also, I might be misunderstanding how the find my protocol works, but how can you send a BLE message to a device you are nowhere near?

You can't, ringing the device is not possible when you're not physically close to it.

jekkos, 2025-11-17:
Any idea if this can be done with a low level bluetooth app? I found these services being exposed  
[image]

malmeloo, 2025-11-17:
Yes it is possible, AirGuard has support for it. I believe Apple's official Android app does as well.

malmeloo, 2025-12-03:
Quick update: I have been playing around with a very experimental implementation of this feature, but I cannot get it to work reliably. However, I have the suspicion that this might be related to the bluetooth stack on my Linux laptop. I have pushed my experiments to #211 and would appreciate some feedback on how well this works for you guys.

Install it using the following command: `pip install git+https://github.com/malmeloo/FindMy.py#feat/accessory-communication`. The following example script should make your accessory play a sound:

```py
from __future__ import annotations

import argparse
import asyncio
import logging
from pathlib import Path

from findmy import (
    FindMyAccessory,
    KeyPair,
    OfflineFindingScanner,
)

logging.basicConfig(level=logging.INFO)

async def scan(check_key: KeyPair | FindMyAccessory | None = None) -> bool:
    scanner = await OfflineFindingScanner.create()

    print("Scanning for FindMy-devices...")
    print()

    scan_device = None

    async for device in scanner.scan_for(10, extend_timeout=True, print_summary=True):
        if check_key and device.is_from(check_key):
            scan_device = device

            break

    print()
    if scan_device:
        print("Device was found in scan results! :D")

        await asyncio.sleep(1)

        print("playing sound")
        await scan_device.play_sound()
    elif check_key:
        print("Device was not found in scan results... :c")

    return scan_device is not None and check_key is not None

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    group = parser.add_mutually_exclusive_group()
    group.add_argument("--private_key", type=str)
    group.add_argument("--airtag_file", type=Path)
    args = parser.parse_args()

    dev: KeyPair | FindMyAccessory | None = None
    if args.private_key:
        dev = KeyPair.from_b64(args.private_key)
    elif args.airtag_file:
        dev = FindMyAccessory.from_json(args.airtag_file)

    device_found = asyncio.run(scan(dev))

    if device_found and isinstance(dev, FindMyAccessory):
        print("Current scan results were used to align the accessory.")
        print(f'Updated alignment will be saved to "{args.airtag_file}".')
        dev.to_json(args.airtag_file)
```

Run it using `python3 <file>.py --airtag_file <your airtag>.json`. Note that the API may change in the future, this is very much a work in progress.

sekerci9090, 2026-09-18:
Ran into a very concrete real-world instance of this. My setup: a cron job runs collect.py (a short-lived Python process) every 2 minutes, which loads a FindMyAccessory from a JSON file via from_json(), calls fetch_location_history(), and exits — but never called accessory.to_json() afterward to persist the updated alignment_date/alignment_index.

Since each cron run is a fresh process, every single invocation for 10 days straight loaded the same stale alignment (from initial pairing/dump time), computed get_max_index() from that old anchor + elapsed wall-clock time, and kept returning the same handful of cached reports whenever the accessory moved somewhere with different Find My network coverage than usual. It looked like the tracker was "working" (no errors, consistent report counts every run) while silently never advancing past a fixed point for hours. Persisting accessory.to_json(path) after every fetch fixed it immediately.

This is a pretty easy trap to fall into for anyone building a polling/cron-based tracker rather than a long-lived in-memory scanning process (the case #262 addresses). Might be worth calling out explicitly in the fetch_location_history() docstring and/or the real_airtag.py example — something like "if your process doesn't stay resident, you MUST persist to_json() after every fetch or your alignment will never advance."