malmeloo/FindMy.py #90: "detect OF devices in Nearby state" doesn't seem to work anymore

thisiscam, 2024-12-04:
Recently the 5 bytes matching trick discussed in #46 appears to stop working. 
I also just tried #50 and seems like I can't find devices that are in nearby state using the example device_scanner.py

malmeloo, 2024-12-04:
Hmm that's strange, I recently used the scanner for another project of mine and could've sworn it received multiple "nearby" reports. That said, I have received a report from someone else saying that the accessory key generation is not working properly, which might be related to your problem. I've been quite busy but will investigate a bit more soon, will keep you posted.

thisiscam, 2024-12-04:
> Hmm that's strange, I recently used the scanner for another project of mine and could've sworn it received multiple "nearby" reports. That said, I have received a report from someone else saying [...]

To clarify, I can see there are nearby devices, but I can't match them with my own devices with known private keys.

malmeloo, 2024-12-04:
Ahh alright, thanks for the clarification. I'll look into it!

malmeloo, 2024-12-05:
Could you try whether #91 fixes your issue? You can install it as a drop-in replacement using `pip install -U git+https://github.com/malmeloo/FindMy.py@fix/nearby-device-discovery`.

thisiscam, 2024-12-06:
> Could you try whether #91 fixes your issue? You can install it as a drop-in replacement using `pip install -U git+https://github.com/malmeloo/FindMy.py@fix/nearby-device-discovery`.

Unfortunately that didn't fix the problem. I was already doing a timeshift adjustment in my own code. Is it possible that Apple changed their "5 matching bytes" protocol?

thisiscam, 2024-12-06:
Some possibilities:
1) The rotating key algorithm was changed
2) The 5 matching bytes protocol was changed
3) I'm not using the correct decrypted plist?? 

Reason 3) is possible is that one of my Mac was updated to macOS 15 recently and I can no longer decrypt the plist on that machine. Ref: https://github.com/MartinPham/FindMySync/issues/38  
So I ended up using an older mac that's still on macOS 13 to decrypt the plist on that machine. Interestingly, I found this new decrypted plist file to use the same private key as the one I had been using before, prior to the macOS15 update. So maybe this explains away the possibility of 3) a little bit, but I'm not sure if there are catches.

malmeloo, 2024-12-06:
That's really strange, and I'm honestly not sure what's going wrong. Of course it's possible that Apple changed the spec, but that's assuming that they somehow have the ability to update both airtags and 3rd party tags over the air.

I'm picking up my own 3rd party tag soon to do some experiments on this.

thisiscam, 2024-12-06:
I'm leaning towards 1) at the moment. If I put my phone on flight mode, I get one (new) separated device, but it also doesn't match with any public key in the time shift range.

thisiscam, 2024-12-06:
Ok this should eliminate possibility of 2):

I put my phone flight mode, got one separated device:
```
SEPARATED Device - DC:B0:6B:C1:E1:20
  Public key:   XLBrweEg0MRQwAt5gk0zk0rRDOQtdW2vr1WCqw==
  Lookup key:   IcctnBNQnY0H3Pqq7JQzGtIQxDNvFAs2yWb2TlH3lkw=
  Status byte:  0
  Hint byte:    0
  Extra data:
``` 
I checked that 
```python
>>> import base64
>>> s = "XLBrweEg0MRQwAt5gk0zk0rRDOQtdW2vr1WCqw=="
>>> base64.b64decode(s)[1:6].hex()
'b06bc1e120'
```

So it seems like the 5 bytes trick is still valid!

It then seems likely that they changed the rotating key algorithm. They _could_ be changing only the official airtag's algorithm afaik. I can see an airtag software version number on my phone --- not sure if recently got updated though!

malmeloo, 2024-12-06:
The public key generated by the library is partially derived from those bytes in the mac address, so I'm not surprised it checks out ;-). And that's interesting, though I guess it makes sense that Apple would want to update the firmware on their own tags. Theoretically speaking it would be possible for Apple to use a new, updated algorithm only on their devices / AirTags, since owner devices are aware of the type of accessory that they're tracking.

If possible, could you try to re-pair the tag, then dump the plist again and see if that fixes the issue, at least temporarily?

thisiscam, 2024-12-06:
> The public key generated by the library is partially derived from those bytes in the mac address, so I'm not surprised it checks out ;-). And that's interesting, though I guess it makes sense that [...]

Yeah, I repaired, dumped and decoded the plist on macOS13, and that did not fix the issue.

malmeloo, 2024-12-06:
That unfortunate; I'll probably pick up a tag next week and experiment a bit on my own. Will let you know how it goes.

thisiscam, 2025-01-06:
Sorry for the ping --- is there an update to this? Should I an opensource tag instead?

thisiscam, 2025-01-08:
Thanks for the update! I am fairly certain that the BLE nearby device tracking functionality is not working for official airtag and official clones anymore. In the mean time while you look at this, do you have any recommendation for a open source drop in replacement tag? As in, do you happen to know (to a good degree of certainty) that particular hardware should work with the BLE nearby device tracking?

malmeloo, 2025-01-08:
The BLE device tracking should work with any "static" OpenHaystack tag. There are a bunch of people working on different firmwares for different tags, so it mostly depends on which hardware you can get your hands on. I am personally using [these](https://nl.aliexpress.com/item/1005007854298968.html) NRF52810 tags with [this](https://github.com/acalatrava/openhaystack_moko) firmware, but there are likely cheaper options available.

kimi4422, 2025-01-09:
> 嗯，这很奇怪，我最近将扫描仪用于我的另一个项目，并且可以肯定它收到了多个“附近”报告。话虽如此，我收到了其他人的报告，称附件密钥生成无法正常工作，这可能与您的问题有关。我一直很忙，但很快就会进行更多调查，并随时向您汇报。

Brother, I have some problems that need your help. Please leave your contact information and I can pay you.

malmeloo, 2025-01-09:
> > 嗯，这很奇怪，我最近将扫描仪用于我的另一个项目，并且可以肯定它收到了多个“附近”报告。话虽如此，我收到了其他人的报告，称附件密钥生成无法正常工作，这可能与您的问题有关。我一直很忙，但很快就会进行更多调查，并随时向您汇报。
> 
> Brother, I have some problems that need your help. Please leave your contact information and I can pay you.

If it's related to this issue then please leave your question here, that way everyone can benefit from the shared knowledge. If it's something really private you want to share, feel free to use my git email.

thisiscam, 2025-01-10:
Just an update on my end: I actually tried to pair my official airtag with an iOS 17 device, and got the plist out from a macOS 13 --- it still does not work. At this point, the only possibility is that they changed the key rotation.

thisiscam, 2025-01-10:
> Just an update on my end: I actually tried to pair my official airtag with an iOS 17 device, and got the plist out from a macOS 13 --- it still does not work. At this point, the only possibility is that they changed the key rotation.

Actually, I spoke too soon! For the above comment, I was testing iOS 17 + macOS 13 with the latest version of the FindMy.py library, which wasn't working. 

However, after switching back to FindMy.py version 0.75 (prior to the nearby scan was implemented) and using my old code from #46, everything is working again! This suggests that an issue in the current FindMy.py scanner code. Upon inspecting the scanner class, I couldn't find the "five consecutive bytes check" in the code.

Since I am able to get everything to work again using iOS 17 + macOS 13 combo, it appears that Apple hasn't changed the AirTag key rotation algorithm itself. But, I am likely using the wrong master key after upgrading to iOS 18 + macOS 15, since the new keychain app does not allow easy extraction of the BeaconStore key to decode the plists, and I had to use macOS 13 to do the plist decoding! Interestingly, if I pair the official AirTag with iOS 18 but extract the plist using macOS 13, I get a plist file that still allows querying the AirTag's location from Apple's server. However, it’s possible that Apple is concealing the true private key with the iOS18 + macOS13 combo.

malmeloo, 2025-01-10:
That's great news, thanks for investigating! So, just to summarize:

- Nearby scanning needs a fix in the library
- iOS 17 + macOS 13 works
- iOS xx + macOS 15 does not work, because the keys can't be dumped
- iOS 18 + macOS 13 does not work when scanning, but fetching locations does work

Correct?

thisiscam, 2025-01-10:
> That's great news, thanks for investigating! So, just to summarize:
> 
>     * Nearby scanning needs a fix in the library

Right. The fix is in #97 

>     * iOS 17 + macOS 13 works
> 
>     * iOS xx + macOS 15 does not work, because the keys can't be dumped
> 
>     * iOS 18 + macOS 13 does not work when scanning, but fetching locations does work
> 
> 
> Correct?

Correct.

malmeloo, 2025-01-11:
Cool, thanks for confirming. I'll look at your PR tomorrow when I have my laptop available.

I think it might be time to seriously start looking into how keychain syncing works. That way we should able to obtain the keys directly without having to dump them from a mac.

thisiscam, 2025-01-11:
> Cool, thanks for confirming. I'll look at your PR tomorrow when I have my laptop available.
> 
> I think it might be time to seriously start looking into how keychain syncing works. That way we should able to obtain the keys directly without having to dump them from a mac.

Keychain (as in the macOS app) syncing? Or do you simply mean how the airtag communicate with an iOS device to decide on the master private key? I do suspect that we can potentially have an middle man when the airtag pairs for the first time. Not sure how feasible that is though!

malmeloo, 2025-01-11:
Setting up an AirTag still requires communication with Apple servers for registration, and I'd much prefer trying to reverse engineer a service running on macOS than one that runs exclusively on iOS 😅. I also think that keychain syncing would be more useful in general, because it would allow people to use AirTags without having to reset and pair them through this library. That said, both are interesting to look into.

thisiscam, 2025-01-12:
> Setting up an AirTag still requires communication with Apple servers for registration, and I'd much prefer trying to reverse engineer a service running on macOS than one that runs exclusively on iOS [...]

I'm lacking knowledge of what is "keychain syncing"?

malmeloo, 2025-01-12:
Keychain as in the icloud keychain that stores secrets in your account. Since the AirTag master secrets are stored in the keychain, that should make it possible to pull them directly from an Apple account after logging in, instead of having to dump them from a mac.

malmeloo, 2025-01-27:
I am still looking into this, as my hackintosh still refuses to sync for whatever reason. To the best of my knowledge people have not yet been able to dump the master secrets from macOS 15, so if you don't have access to a < 15 machine it's going to be a challenge.

malmeloo, 2025-02-03:
(continuation from #31) @aircable: Do you have the exact pairing date of your tag for me? It should also be in the plist, but it appears to be missing for some reason. Did you edit that part out or is it missing from the original file?

You mentioned that your tag was paired in 2023, but I was able to find the private key belonging to the scan you posted at index ~13291. That's roughly 13291 * 15 minutes = 4.5 months after pairing time, so it can't have been in 2023, unless there's an issue with the library.

If I select September 12th 2024 as pairing date and use the values from the plist you posted, I can pull location reports from Apple that were generated earlier today.

aircable, 2025-02-03:
> (continuation from [#31](https://github.com/malmeloo/FindMy.py/issues/31)) [@aircable](https://github.com/aircable): Do you have the exact pairing date of your tag for me? It should also be in the [...]
        <key>pairingDate</key>
        <date>2023-08-31T18:32:21Z</date>

> You mentioned that your tag was paired in 2023, but I was able to find the private key belonging to the scan you posted at index ~13291. That's roughly 13291 * 15 minutes = 4.5 months after pairing time, so it can't have been in 2023, unless there's an issue with the library.
> 
How did you do that?

> If I select September 12th 2024 as pairing date and use the values from the plist you posted, I can pull location reports from Apple that were generated earlier today.
Like this, from Items.data:
        <key>signature</key>
        <data>
        1jBBM7vhRl8/fQtoasef9yAPCS6d0JdFDuTpnTeHk9BWpekPBSi6PkBAUl1lGAQiL6p+
        AqIwqbujoCc1Zy8MCQ==
        </data>

What app do you use to decrypt that?

aircable, 2025-02-03:
I did mention that the tag was off for an extensive amount of time. That's why you could not find it. But tell me, the analyze_key.py file I shared did not go to index 31291?
Here is the location cache from Feb 3: 15:15, last location 21h ago
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>identifier</key>
	<string>F61337FA-3307-450E-8103-D2ADE0708699</string>
	<key>secureLocation</key>
	<dict>
		<key>altitude</key>
		<real>0.0</real>
		<key>course</key>
		<real>-1</real>
		<key>findMyId</key>
		<string>MTAxOTIwNDg1Mw~~</string>
		<key>floor</key>
		<string>$null</string>
		<key>horizontalAccuracy</key>
		<real>65</real>
		<key>latitude</key>
		<real>36.992891175123489</real>
		<key>locationLabel</key>
		<string>$null</string>
		<key>longitude</key>
		<real>-121.97523331689773</real>
		<key>motionActivityState</key>
		<integer>0</integer>
		<key>publishReason</key>
		<integer>100</integer>
		<key>speed</key>
		<real>-1</real>
		<key>timestamp</key>
		<date>2023-08-20T05:30:43Z</date>
		<key>verticalAccuracy</key>
		<real>0.0</real>
	</dict>
</dict>
</plist>

aircable, 2025-02-03:
finding more information, from OwnedDeviceKeyRecord. The deviceIdentifier is the one for that Airtag
```
	<key>deviceIdentifier</key>
	<string>1E257350-56A1-4347-8953-B37344888AC2</string>
	<key>identifier</key>
	<string>1FFB4B8E-E6B9-4DDB-A06C-45FC2638F2A1</string>
	<key>privateKey</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			BFrpmM2QZJUEjiGUM52ejt5AshEbHjWmPgzjP47+2p2NpbnaYS20
			7SGDgseqoWzL2DroseLl0r6O4miDNGNAvX54xNP/HNZDrsT/1FCx
			JsFvZOHwYg==
			</data>
		</dict>
	</dict>
	<key>publicKey</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			BFrpmM2QZJUEjiGUM52ejt5AshEbHjWmPgzjP47+2p2NpbnaYS20
			7SGDgseqoWzL2DroseLl0r6O
			</data>
		</dict>
	</dict>
```

malmeloo, 2025-02-05:
Sorry, missed your responses.

> I have, my Macbook Air M1 is Sonoma 14.4. Let me know what to do.

My plan was to pair a tag and obtain the very first key it broadcasts, then start from there. That should allow me to verify the assumptions about timezones etc. as well, since I'm not 100% sure that's correct yet (even though it shouldn't cause any issues in practice)

Thanks for the pairing date! Do think it'd be possible to connect the tag to an owner device to make it update its internal clock? I'm curious to see if the key rotation algorithm would immediately jump to the "correct" key or not.

> How did you do that?
> ...
> What app do you use to decrypt that?

I used the same debugging script you posted but set the pairing date all the way back to 2018 or so. Then filled in the public key (not the lookup key) from your scan and let it run until it found a match. The date of the match is wrong but relative to the pairing date that is set, and since it was ~4.5 months after the pairing date I used, I figured it must've been paired ~4.5 months ago. But I forgot about your tag being turned off for a while, so if it was paired in 2023 that still checks out.

Using the above method to find the "real" (wrong in theory but real according to the accessory's internal clock) pairing date, I plugged it into the `real_airtag.py` example to obtain the location reports.

I'm not sure what those SecureLocationInfo and OwnedBeacon files are supposed to represent. The private/public key lengths don't match up with the EC keys used in FindMy. However, the following strings look interesting:

```
<key>ownedDevicePublishToACSN</key>
<true/>
<key>ownedDevicePublishToACSNWithVersionCheck</key>
<true/>
<key>ownedDeviceStopPublishToACSNIOSVersion</key>
<string>16.0</string>
<key>ownedDeviceStopPublishToACSNMacOSVersion</key>
<string>13.0</string>
<key>ownedDeviceStopPublishToACSNWatchOSVersion</key>
<string>9.0</string>
```

ACSN (anonymous crowd-sourced network) is the server-sided part of FindMy running at Apple. It looks like there are lower OS version bounds for when to enable participation in the network? I don't think it's related to this issue, but interesting nonetheless.

aircable, 2025-02-05:
> I used the same debugging script you posted but set the pairing date all the way back to 2018 or so. Then filled in the public key (not the lookup key) from your scan and let it run until it found a [...]

So, basically you changed the airtag plist file and put some different numbers in and just ran real_airtag.py airtag.plist, right? What did you change in the file, can you please post? I just want to make sure I understand what you did.

malmeloo, 2025-02-06:
I left the plist intact, but manually set the pairingData (since it was missing). I've filled in your details in the files below.

Run this with the path to the plist to find the private key:
[airtag_tester.txt](https://github.com/user-attachments/files/18689990/airtag_tester.txt)
It finds an approximate time which is roughly 4.5 months after the pairing date I set in the file.

I then played around with the pairing date around september last year until it found the most recent location reports:
[real_airtag.txt](https://github.com/user-attachments/files/18690002/real_airtag.txt)

aircable, 2025-02-07:
Thank you so much, I was able to follow your steps. Got data.

But not those steps:
> I used the same debugging script you posted but set the pairing date all the way back to 2018 or so. Then filled in the public key (not the lookup key) from your scan and let it run until it found a [...]

Using the airtag_testerl I find the key at 
  - Approx. Time:  2023-06-25 12:00:00+00:00
  - Type:          KeyType.SECONDARY
 
That is vs the reported pairing time:
	<key>pairingDate</key>
	<date>2023-08-31T18:32:21Z</date>
2 months earlier.

Then, as you did, searching for the data using datetime(2024, 9, 11) leads to the most keys found. But no other date determined by the airtag_tester script.

My interpretations maybe is that the possible_keys you generate are just not enough. How many are you actually generating?
I found that querying the server, I find 7 keys and they rotate about once a day.

I also found out that a span of 4 days will give you the maximum number of reports, in out case 2024/9/10-13.
The dates that gives at least one key are 14 days: 24/9/4 and 24/9/18. As far as I remember that is the maximum time the server stores keys anyway. That makes sense then.

Do those numbers ring a bell? Are you creating a 4 day window with the key generator?
My question is, what is the number of days the key rotate, starting from the beginning.  
I could not find a repeat. Does that mean there is only one possible pairingDate that is successful?

aircable, 2025-02-07:
I have actually verified that the pairing date for that airtag was indeed 2023/8/31. The plist file is correct.
The difference is 2024-9-11 (real_airtag) vs. 2023-8-31 (plist).
Removing the battery does de-sync the clock then, I guess. So, why does the FindMy app locate it anyway?
Checking way more keys?

malmeloo, 2025-02-07:
Let me try to explain the AirTag's inner workings in a bit more detail. I hope that clarifies things a bit, while at the same time serving as an attempt at more understandable public documentation :-). I'll try to elaborate on (almost) everything start to finish, so it might include some redundant information.

As you are probably aware of by now, the FindMy network works using cryptographic key pairs. The public key is broadcasted by a device, and we use the private key to obtain the device's location. For custom tags, we almost always use static keys, meaning that the key broadcasted by our custom device does not change. This means that the private key is very predictable (it's always the same!), so fetching locations from Apple is very easy. However, AirTags and compatible 3rd-party accessories rotate their keys in an almost unpredictable way in order to increase privacy (+ obfuscation??). While we can use the same method to obtain an AirTag's location as we use for our custom tags, we also need to know which keys the AirTag has broadcasted during the time range that we would like to query. Otherwise, we'd just be throwing a LOT of random keys at Apple's servers in the hope that some of them will work. That's just inefficient and very slow.

AirTags have two distinct key derivation algorithms. The exact cryptographic details of how these work does not matter, but what does matter is that each of these algorithms outputs a single key for a certain value $i$ or $j$ ("index"). One of the algorithms generates a primary key $K_{p,i}$, while the other generates a secondary key $K_{s,j}$. We also assume it to be nontrivial ("impossible") to find these key(s) for a certain index without knowing certain shared secrets, which are stored in the plist file. Since the shared secrets are stored in the plist, obtaining this file would allow us to compute the keys that an AirTag would generate. The only additional thing we need to know is the (approximate) value of $i$ or $j$ so that we do not need to fully exhaust the output range of the key derivators (this is theoretically infinite, since $i$ and $j$ can be any natural number). It is worth noting that, while $i$ and $j$ can be equal, they usually aren't. And even if they were, the resulting primary and secondary keys would be different since the key derivators are seeded with different values.

The main issue is actually finding $i$ and $j$. AirTags use only one of its two key derivators when selecting a new key to broadcast. $i$ and $j$ are selected based on how long ago the tag was paired, while the chosen key derivator depends on the current state of the tag (connected, nearby or separated). The tag is "connected" when actively or very recently in range of its owner device, "nearby" when the tag disconnects from its owner device, and "separated" when it has been disconnected for a longer period of time. The tag is also in "separated" mode after a power cycle.

When the tag has recently actively connected to its owner device, it broadcasts a "lite" version of the regular FindMy broadcast data. This data is not enough for other devices to identify and upload its location, but is sufficient for owner devices to recognize it as one of their own. They do this by only broadcasting part of the **primary** key. This key changes by incrementing $i$ every 15 minutes and deriving the new key.

When in "nearby" state, the tag also broadcasts its **primary** key, but in full this time. This "enables" the stranger-finds-my-tag features of FindMy. Again, this key changes every 15 minutes, because $i$ is incremented every 15 minutes. Interesting to note here is that this is why the tag's unwanted tracking detection features are "disabled" when it only recently disconnected from its owner device: the tag essentially changes its identity every 15 minutes, so nearby iPhones do not detect it as the same tag that is following them.

When in "separated" mode, the tag starts broadcasting its **secondary** key. This key only changes once per day. Note that even while in separated mode, **$i$ will keep incrementing by 1 every 15 minutes.** When transitioning into separated mode, $j$ is recomputed as $j = i/96 + 1$. When 4 AM hits and the tag is **still** in separated mode, we increment $j$ by 1. Note that, since we do integer division and there are exactly 96 15-minute "blocks" in a day, $j$ is essentially the same as what $i$ would have been at the "last" 4 AM.

The annoying part is that we are essentially operating in the dark: we **do not know** which state the tag is in, or what the "last" value of $i$ and $j$ were. Apple has a "strategic" edge over this library because they can track that information and share it among their devices. However, we can still make an educated guess on what the approximate values of $i$ and $j$ are, and add a margin (~24 hours) when calculating the keys so that the probability of "missing" a key is quite low (under normal circumstances, more on that later).

The above means that at any arbitrary point in time after the pairing date, we have two or three potential keys that the tag could be using:
1. The primary key of the current 15-minute block, if it is in connected or nearby mode;
2. The secondary key computed from $j = i/96 + 1$, if the accessory has been power cycled or went into separated mode earlier today;
[... cut ...] could theoretically look at the times corresponding to the location reports of a single key, and adjust $i$ based on that. So if we believe that a certain key A should be generated for $i = 100$ at 5 PM, but the location reports show that A was seen at 3:30 PM, we know that $i$ should be decremented by 6. However, it would be up the the users of the library to do something with this information and store it for later.

I do realize this is a lot of text, but hopefully this clarifies everything 🙂

aircable, 2025-02-09:
Wow, thank you so much. Your description is incredible. I believe it will be very helpful, for others too.
It is surprising to me that the key rotation for the real airtags create so many keys, nearly infinite since we don't know the range of i or j.
For us, only the separated state is important, since we are looking to track lost airtags. And it could have been powered off for some time. 

So, I worked on the airtag_tester a bit and created a program that counts backwards and tries to find the correct key as well as the computed number of days offset, which then leads to the corrected pairing date you can use in your real_airtag script.

And it worked. Output:
KEY FOUND!!
KEEP THE BELOW KEY SECRET! IT CAN BE USED TO RETRIEVE THE DEVICE'S LOCATION!
  - Key:           VblRR9fkMzLo0VjJGAct2JSMU4Vk5Jm+wMOdpQ==
  - Approx. Time:  2024-01-29 18:15:00+00:00
  - Type:          KeyType.SECONDARY
Computed Offset: 376 days
Corrected Pairing Date: 2024-09-10 18:32:21+00:00

Which is exactly the date you would use to find the real Airtags.

Maybe we should now integrate the airtag-decryptor-3.swift, the aitrag_tester_2.py and the real_airtag_2.py scripts into just one "get location from real airtags".

[airtag_tester_2.txt](https://github.com/user-attachments/files/18725937/airtag_tester_2.txt)

malmeloo, 2025-02-10:
Nice! I don't think the scripts would need to be fully integrated though, since "calibration" could be a one-time thing. Once we get at least one location report we can use that to do further offset calibration. I'm thinking of something similar to the following workflow:

1. Obtain the plist
2. Use a combination of the scanner script and airtag_tester to "calibrate" and find the initial offset
3. Automatically calibrate after every fetch of location reports and allow that calibration to be saved somehow

If we make the FindMyAccessory class serializable, the offset could be stored in the JSON representation, and calibration after that can be done without referring to the plist.

aircable, 2025-02-10:
Sounds about right, which means that the library need to track i and j. 
Would you change the findmy library or do that somewhere else?

malmeloo, 2025-02-10:
Probably in the library. We somehow need to keep track of the expected generation time for each key, then if it does not align with the time of the reports for that key we can derive a new offset from the difference in those times. It also doesn't need to be very accurate as long as there's a 24h margin.

alfs, 2025-02-22:
Thank you @malmeloo for the very detailed explanation.

I have this exact problem with two dozen tags being paired several months ago. I re-inserted the battery saver tab after pairing. Now when tags are powered and field deployed (separated mode), tag key indexes are off compared to the time-window calculated by FindMy.py.

I presume the tag clocks may also naturally drift due to battery voltage dropping (this would be interesting to investigate by itself, and could be quantified by observing a faraday-caged tag and ble sniffer for some time).
Over time, as batteries expire and gets replaced, tags will then also lose 15 minute time synchronization from pairing date/time.

I therefore think your suggestion (3) with automatic continous calibration towards reported time, and saving of estimated i,j indexes, is a necessity for providing reliable lookups and also has a benefit reducing the report query load.

Key question: **would an key-index-centric design make more sense than a time-based design?** 

Idea: after initial calibration (based on time, opportunistic probing or full probing), each invocation of FindMy.py would
 a) save the clock time it was last run
 b) Calculate n, m from hours, days of (now() - last_invocation), and request reports for {i .. i+n}, {j .. j+m}. To account for tag clock skew, add a suitable number (10? does not need to be large, over time it will adjust as long as findmy is run periodically vs possible clock skew).
 c) update cached i, j values for the latest successfully received reports

If FindMy.py is run at least once per week, it will in the worst case be the same as the current "fetch_last_reports" that queries 7*24 hours + 12+12 hours = 768 id queries only based on time.
However if it's run once per day, it will only send queries for the it's that are actually possible in that time frame, and would not need extra +-12h marigins (which is a magic number I guess, skew can be outside this window).

If FindMy.py has not been run for a long time (say months), there can be some opportunistic/sampling discovery, for example instead of 96 reports per day, try 10 reports per day and re-synchronize after an initial round.

This method requires a local state to be kept and updated, but I think the value far outweighs the drawbacks - and the time-based algorithm can still be used of course, if the user chooses. In my use-case with many severely de-synchronized tags, a key index offset is a necessity.

malmeloo, 2025-02-22:
Yeah I've been thinking about this a bit more recently and I'm increasingly less opposed to tracking internal state for airtags. The library already has an abstract base class for things that need proper state handling, so we could just extend that to the accessory class. Maybe with a warning if it gets garbage collected with unsaved runtime data, to really get the message home.

I think the final implementation will probably look similar to what you're proposing. Keep track of the last fetch time and synchronized i and j, then calculate offsets from there. Your idea about opportunistic discovery is quite interesting, but I think we can actually reduce the number of keys per day to only 1 for this. Since the vast majority of location reports will likely originate from secondary keys, and those only regenerate once per day. However, that does mean that we will still need that 12-24 hour margin, because we cannot recalibrate the indices super accurately using those secondary keys. After all, they're valid for 24 hours instead of 15 minutes.

alfs, 2025-02-23:
I did a PoC (attached) that scans secondary keys for 200 days and saves the last key index for the found report in a cache json file, reused at next run.

It now finds my missing tags (great!), but there is still some problem with reusing indexes (seems keys_between don't produce the same result with a full-range scan [1, 200*24*60/15 ] or a subset scan. I'll look into this a bit more.

There is also some other indexing problem, where keys with higher index has lower timestamps. 

For example:
```
Index 11903 Key KeyPair(name="None", public_key="***", type=KeyType.SECONDARY)
 - KeyReport(hashed_adv_key=***, timestamp=2025-02-23 04:59:39+01:00, lat=123.3756623, lon=123.4780003)
...

Index 13546 Key KeyPair(name="None", public_key="***", type=KeyType.SECONDARY)
 - KeyReport(hashed_adv_key=***, timestamp=2025-02-16 20:40:13+01:00, lat=123.3755094, lon=123.477932)
```

The indexes are numbers by iteration from keys_between - perhaps these are unordered?

```
   for k in airtag.keys_between(firstindex, lastindex):
        # Only use secondary keys that rotate daily to reduce lookups
        if k.key_type == keys.KeyType.SECONDARY:
            kdict[k] = i
            klist.append(k)
        i += 1
```

I'm surprised of the high index numbers vs pairing date.
Index 13546 / 96 = 130, which should correspond to 130 days? The tags were joined in december 2024, 77 days ago, and were also powered off around 50 days.

Anyway, this approach seem promising for very efficient lookup and responses, which enables lookup scaling of many tags (after calibration, 1-2 lookups needed per tag that can be packed in 256 id chunks) vs quering several hundred id's per tag and getting possibly thousand of reports back.

[calibrate.txt](https://github.com/user-attachments/files/18932942/calibrate.txt)

alfs, 2025-02-23:
I updated the the indexing code (iterating over keys_at()), and it's now consistent from initial calibration to subsequent runs. Index now matches expected operating time (key index of the last successful report is 1055 => ca 11 days of operation which is according to expectations (paired in december, powered off, and deployed 1,5 week ago.

[calibrate.txt](https://github.com/user-attachments/files/18933675/calibrate.txt)

thisiscam, 2025-02-24:
> I did a PoC (attached) that scans secondary keys for 200 days and saves the last key index for the found report in a cache json file, reused at next run.
> 
> It now finds my missing tags (great!), but there is still some problem with reusing indexes (seems keys_between don't produce the same result with a full-range scan [1, 200*24*60/15 ] or a subset scan. I'll look into this a bit more.
> 
> _edit_ - keys_between() returns a set, so indexing can't be done this way.
> 
> ```
>     def keys_between(self, start: int | datetime, end: int | datetime) -> set[KeyPair]:
>         """Generate potential key(s) occurring between two indices or timestamps."""
>         keys: set[KeyPair] = set()
> ```
> 
> Either the library can use a sequence, or I can just do the same iteration as the library does ( while start < end: keys.update(self.keys_at(start)) ) and use a list instead.
> 
> Anyway, this approach seem promising for very efficient lookup and responses, which enables lookup scaling of many tags (after calibration, 1-2 lookups needed per tag that can be packed in 256 id chunks) vs quering several hundred id's per tag and getting possibly thousand of reports back.
> 
> [calibrate.txt](https://github.com/user-attachments/files/18932942/calibrate.txt)

You could perhaps use https://pypi.org/project/ordered-set/

malmeloo, 2025-02-24:
> I did a PoC (attached) that scans secondary keys for 200 days and saves the last key index for the found report in a cache json file, reused at next run.
>
> It now finds my missing tags (great!), but [...]

Nice! The use of sets is indeed a bit of an issue, but once it's implemented in the library that shouldn't be an issue anymore. I'll see if I can find some time this week to create a first version with the fix.

parawanderer, 2025-03-01:
Hi so I was curious about this project as I had heard about the Apple AirTags being the best on the market in terms of the network (with all Apple devices apparently being opt-in by default lol) and wanted to get in on this as a non-Apple-user. My end-goal is to be able to interactively view the AirTags on any arbitrary device, particularly on Android (because screw walled gardens lol).

So first of all, thank you for the neat project! 

So at this point I got this far:

1. I have a new set of AirTags that I registered using a new iPad. Created a new Apple account for the iPad. The AirTags showed up in the `FindMy` app on the iPad and I customised them on there by giving them some custom names.
2. I then set up an `anisette server` as in [dchristl/macless-haystack](https://github.com/dchristl/macless-haystack?tab=readme-ov-file#server-setup) (step 1 and 2 only)
3. I next set up a `MacOS Sonoma (14)` VM via this project: [sickcodes/Docker-OSX](https://github.com/sickcodes/Docker-OSX) using Windows 11 + wsl + Docker. This took me a few hours to get through the OS installer, but I basically got away with copy pasting the suggested docker run command and just clicking around in the installer UI a bit. 
4. I then set up the MacOS vm with the same Apple account as the iPad, and enabled the location-related stuff when prompted. I opened the `FindMy` app on the vm and saw the AirTags shown in there too (albeit without the map's background or icons showing up for whatever reason, but the left sidebar had the AirTags under the "Items" tab).
5. I continued to using the first Swift script linked in [this comment](https://github.com/malmeloo/FindMy.py/issues/31#issuecomment-2088586978) to export the `plist`s to the location mentioned by the script (`/tmp/com.apple.icloud.searchpartyd`) by just running:    
    ```bash
    swift airtag-decryptor.swift
    ```
    This worked for me without any of the `incorrectKeysize` errors from the previous thread. Though I realise now that I'm not sure how I got to the exact file locations for these `plist` outputs anymore as they're actually at `/private/var/folders` for me...
 6. I then exported a bunch of the `plist` files that it generated at the output path to iCloud for the Apple account and retrieved those back on my Windows host for my own further inspection and editing.
 7. I understood that the files in `OwnedBeacons` are the `plist` files that are needed to get the location data for the keys, that is the argument you need to provide to the example script provided in [`examples/real_airtag.py`](https://github.com/malmeloo/FindMy.py/blob/main/examples/real_airtag.py). 
    
    The file name of the files `OwnedBeacons/<name>.plist` matches to the `BeaconNamingRecord/<name>/<something-else>.plist` and the (single?) latter file contains the customised AirTag names that I set up in the iPad.

    Running the example [`examples/real_airtag.py`](https://github.com/malmeloo/FindMy.py/blob/main/examples/real_airtag.py) against any of the AirTags seems to produce a working list of history results for the 2 AirTags I tested immediately. But not for the ones I tested an hour or so later.

Supposing I had a constantly on MacOS vm as in this experiment.  These questions remain from my side: 

1. Is what I am seeing with the other AirTags here indicative of the same issue previously reported in this thread? Do they stop updating?
2. Do I need to refresh the `plist` files regularly? If yes, when do I need to refresh them? What about the customisations such as name changes?
3. Is it even feasible to set up some automation around any of this at this point in time? 
4. How often does the login step for the account need to be redone (particularly that 2FA auth step)? 
5. How breakage-prone would automation around things like file formats/Airtags be (ignoring the whole vm and `plist` re-fetching)?

This might be a lot, so regardless thank you in advance!

malmeloo, 2025-03-01:
1. Likely, yes, but you're the only one who can 100% verify that it is indeed the case. A couple messages up there is a script that you could use. If it works, it's the same known issue.
2. It should not be necessary to refresh the plist. The only thing that matters is the encryption keys, and they are set once at pair time. However, if you re-pair your Airtag, you will need to get the plist again.
3. Is it feasible? Yes, but it requires some manual know-how to work around the current issue. You might be able to piece something together from the resources in this thread, but if not, wait until the fix lands in the library.
4. Theoretically never, but you must make sure that you always use the same anisette server and that the server does not lose its configuration data.
5. With breakage-prone do you mean Apple breaking this? The existence of 3rd-party tags fossilizes the protocol somewhat, but in the end we're still playing in Apple's walled garden, and they could (attempt to) kick us out at any time. If you're automating requests then stick to 15-30 min fetch interval minimum, and keep the request time range as short as possible. Just don't blame me if they ban your account - use a burner account if necessary.

parawanderer, 2025-03-02:
Awesome, thanks for the reply and clarifications, and for the great project! 🙏

wilkyconsultants, 2025-03-09:
Interesting project for sure! 

Sorry, please remove if I am off topic here but kinda need some macos airtag help!

My research is still at the stage where I use the FindMy cache files to pull the airtag data. Anyone know how to properly get the key required to decrypt the ~/Library/Caches/com.apple.findmy.fmipcore/Items.data file in Sonoma 14.4++? 

FYI, I was able to see the BeaconStore on macos 14.7 using :  

security find-generic-password -l "BeaconStore" -g 

but the password supplied does not work using AES GCM decryption of the plist seems.

ie. 
plutil -convert xml1 -o Items.data.test.xml Items.data
./de_aes_gcm.py Items.data.test.xml KEY_RECEIVED_FROM_THE_SECURITY_COMMAND Items.data.test.output

Results in : error: Decryption failed: MAC check failed
Which basically means the key is wrong..

-- Here is my decrypt code:

cat de_aes_gcm.py 

#!/usr/bin/python3
import json
import base64
import re
import argparse
from Crypto.Cipher import AES

def extract_data_from_plist(plist_file):
    with open(plist_file, "r") as f:
        plist_content = f.read()
    
    encrypted_data_match = re.search(r"<key>encryptedData</key>\s*<data>(.*?)</data>", plist_content, re.S)
    
    if not encrypted_data_match:
        raise ValueError("Encrypted data not found in plist file")
    
    encrypted_data = encrypted_data_match.group(1).replace("\n", "").replace(" ", "")
    return encrypted_data

def decrypt_data(key_hex, encrypted_base64):
    key_data = bytes.fromhex(key_hex)
    encrypted_data = base64.b64decode(encrypted_base64)
    
    nonce = encrypted_data[:12]  # First 12 bytes are the nonce
    print("nonce=",nonce)
    tag = encrypted_data[-16:]   # Last 16 bytes are the tag
    print("tag=",tag)
    ciphertext = encrypted_data[12:-16]  # The rest is the ciphertext
    
    cipher = AES.new(key_data, AES.MODE_GCM, nonce=nonce)
    decrypted_data = cipher.decrypt_and_verify(ciphertext, tag)
    return decrypted_data.decode('utf-8')

def main():
    parser = argparse.ArgumentParser(description="Decrypt AES-GCM encrypted plist data.")
    parser.add_argument("input_file", help="Path to the input plist file")
    parser.add_argument("key", help="Hexadecimal encryption key")
    parser.add_argument("output_file", help="Path to save the decrypted JSON file")
    
    args = parser.parse_args()
    
    encrypted_base64 = extract_data_from_plist(args.input_file)
    decrypted_json_str = decrypt_data(args.key, encrypted_base64)
    
    decrypted_json = json.loads(decrypted_json_str)
    with open(args.output_file, "w") as json_file:
        json.dump(decrypted_json, json_file, indent=4)
    
    print(f"Decryption successful. Output saved to {args.output_file}")

if __name__ == "__main__":
    main()

malmeloo, 2025-03-09:
I really would not recommend building your own decryption program; it's simply not necessary. Use the searchparty-keys program as discussed in #31 instead. If you need more help feel free to leave a comment in that issue.

wilkyconsultants, 2025-03-09:
> I really would not recommend building your own decryption program; it's simply not necessary. Use the searchparty-keys program as discussed in [#31](https://github.com/malmeloo/FindMy.py/issues/31) instead. If you need more help feel free to leave a comment in that issue.
> 
> Edit: I modified your comment to use a code block (using 3 backticks: ```) to make it more readable.

Sorry to reply here but just fyi, I had tried to use the swift code as outlined but it just errored out on illegal instructions etc.. I tried my best to fix it but it now just says: Invalid hex key format. Have no idea about that. 

Seems to try to run this but I did it manually and it does not work:

security find-generic-password -l 'FindMyAccessories' -g
security: SecKeychainSearchCopyNext: The specified item could not be found in the keychain.

Thanks for your help

malmeloo, 2025-03-09:
Unfortunately I am not very well versed in Swift, and obtaining the required plist files has mostly been a community effort. But as I said you're better off posting in #31 than here, as the people subscribed to that issue are probably more likely to be able to help you.

wilkyconsultants, 2025-03-09:
> Unfortunately I am not very well versed in Swift, and obtaining the required plist files has mostly been a community effort. But as I said you're better off posting in [...]

Thanks again, I was actually able to get it running but found that the plist files are way harder to work with than the FindMy CACHE: ie. 

~/Library/Caches/com.apple.findmy.fmipcore/Items.list where everything is conveniently in one json tree. Just need to figure out the key for it and I am all set..In the mean time I stay on Sonoma 14.3.1 where the cache is in the clear. Cheers

parawanderer, 2025-03-20:
Hi, regarding `.plist` file decoding, I have a solution for this in Python that might be more friendly to library users here to adapt to their own needs (I also don't know Swift): [airtag_decryptor.py](https://github.com/parawanderer/OpenTagViewer/tree/main/python#airtag_decryptorpy)

I'm assuming it also only works on MacOS 14 and stopped working on 15, just like the Swift script has, since it's a python port of the same logic in the Swift script.

TaeHagen, 2025-08-29:
To clarify, what is the actual issue here? Is it mismatched key rotations?

I'm looking to implement findmy in OpenBubbles, and want to make sure I do it right and don't have compatibility issues.

I do have access to the plist you are taking about from the cloudkit DB.

TaeHagen, 2025-08-29:
If it's any help, at least on the cloudkit end there is record that keeps track of the current rotation

malmeloo, 2025-09-07:
Good news! AirTag fixes are finally live on the main branch and will be included in next release, which will be out very soon. There are quite some changes both in API design as well as usage, so I'll include a brief 'tutorial' below. These notes will be included in the docs at some point, hopefully before release.

I would love to have some testers to confirm that it works for your accessories as well! If there are any issues, please do let me know. 

## Obtaining accessory JSON
Accessory objects used to be represented as plist files taken directly from a Mac, but this has changed. The new changes require some state to be kept for each accessory. This state is represented on disk as a JSON file containing various attributes, including alignment parameters. You will need to convert your existing plist files to the new JSON format in order to query locations.

#### Option 1: I already have a plist file
If you already have a plist file, use the following new example script to convert to the new format:
```
python3 examples/plist_to_json.py accessory.plist airtag.json --alignment-plist alignment.plist
```

This will convert the `accessory.plist` file into the new JSON format and write the result to the `airtag.json` file. Don't have an `alignment.plist`? You can leave out that parameter and fix the alignment later (see below).

#### Option 2: I have not dumped my accessory yet
You can use the new experimental decryptor built into FindMy.py itself. On a Mac, run the following command:
```py
python3 -m findmy decrypt --out-dir accessories/
```

This will attempt to dump all FindMy accessories as JSON files to the `accessories` directory. Note that you may have to enter your keychain password in the popup that appears; therefore this command may not work over SSH.

It's possible that the command doesn't work on certain versions of MacOS. If this is the case for you, dump the plist files using any method previously discussed and then convert it to JSON using Option 1.

## Fixing key alignment
If you used Option 1 without the `alignment-plist` parameter, you may need to manually align the key generator with your tag. While the library will automatically correct the alignment every time you query your tag's location, the first fetch may cause excessive API requests to Apple's servers if the alignment is very off. You do not need to do this if you did supply the alignment plist or used Option 2 to dump the accessory, as in this case, the alignment parameters will be copied from Apple. However, aligning using one of the methods below never hurts, and can be very easy depending on your situation.

#### Option 1 (preferred): Aligning using Bluetooth
If you are physically close to your tag, aligning is easiest done over Bluetooth. Use the following script to discover nearby devices and attempt to align the tag:
```py
python3 examples/scanner.py --airtag_file path_to_airtag.json
```

If all went well, you will see a message saying `Current scan results were used to align the accessory`.

#### Option 2: Copying alignment parameters from Apple
Simply re-export the accessory using one of the options specified above, making sure to include the `alignment-plist` parameter if you use Option 1.

#### Option 3: Re-pair the accessory
Re-pairing an accessory will reset its key rollover algorithm, making the first fetch much more efficient. However, due to the amount of work involved in doing this and then re-dumping the accessory, I would only do this if your accessory was paired a while ago (let's say >6 months) and the previous two options don't work for you.

#### Option 4 (not recommended): Just send it
The last option would be to 'just send it' and use the `airtag.py` example script to fetch locations, then wait for the library to figure out alignment while hammering Apple's servers. You will most likely have better luck with this if your accessory was paired relatively recently. Please don't use this option unless the other three methods don't work for you, especially if your accessory was paired a long time ago. If you do it regardless and decide to ignore my advice, it's your own fault if Apple bans your account because you were hammering their API.

closed 2025-09-07 by malmeloo