malmeloo/FindMy.py #96: Documentation on Home Assistant integration

cascornelissen, 2025-01-09:
Hi! I've been following this project for a while because I'm interested in having my AirTags available in Home Assistant.

I noticed the latest published release includes _Fix compatibility with latest version of Home Assistant_ in its notes. I noticed there's also a [malmeloo/hass-FindMy](https://github.com/malmeloo/hass-FindMy) repository but its README mentions it doesn't work with official AirTags just yet while the README of this repository mentions supporting them.

Apologies if I missed some documentation somewhere ❤️

malmeloo, 2025-01-09:
Hoi! You didn't miss any documentation - it just hasn't been written yet ;-). The Home Assistant integration depends on this library for the actual FindMy functionality, but also has to implement some features itself, such as the config flow. The library supports locating AirTags and other official or 3rd party accessories (well, *should* support, see #90), but the config flow and other logic to support this has not (yet) been implemented in the integration.

Every once in a while this library needs to be updated to support the latest version of HA, because it shares some dependencies with the core HA python package. If they bump a shared dependency such as `cryptography` to a new minimum version, this library also needs to support it to prevent [these](https://github.com/malmeloo/hass-FindMy/issues/4) issues.

I hope that answers your question 🙂

thisiscam, 2025-01-10:
> Hi! I've been following this project for a while because I'm interested in having my AirTags available in Home Assistant.
>
> I noticed the latest published release includes _Fix compatibility with [...]

I have a homebrew solution that works with official airtags through the MQTT integration. It's a bit messy to setup, but let me know if you are interested

nightah, 2025-01-11:
> I have a homebrew solution that works with official airtags through the MQTT integration. It's a bit messy to setup, but let me know if you are interested

Does it also work with other devices? I certainly would be keen, if so.

thisiscam, 2025-01-11:
> Does it also work with other devices? I certainly would be keen, if so.

You mean other apple devices? Or other openhaystack tags?

thisiscam, 2025-01-11:
@nightah  Here it is: https://github.com/thisiscam/ha_findmy_mqtt 

I'm sure @malmeloo will come up with a much smoother integration in hass-Findmy than this :)

malmeloo, 2025-01-24:
Closing because I think this question should be answered, feel free to reopen if not

closed 2025-01-24 by malmeloo