malmeloo/FindMy.py #98: Invalid OF data length: 21 instead of 0

thisiscam, 2025-01-10:
A minor issue around https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/scanner/scanner.py#L178 

When I use the scanner, I keep getting this error in my log. 
What might be happening? Could it be some non-apple device triggering this?

Skowt, 2025-01-10:
I'm getting `21 instead of 2` 🤔 Seems like the payload might've changed?

thisiscam, 2025-01-10:
To clarify, the scanner still works (it detects my device), but just pops this error message from time to time

Skowt, 2025-01-10:
Yeah scanning works although when I put my airpods and watch into lost mode, it didn't detect them as separated devices.

malmeloo, 2025-01-10:
This is a bit of a strange one, basically the library thinks that the payload type is different than it actually is. Should be quite harmless, but a bug regardless; I'll look into it.

Skowt, 2025-01-11:
> This is a bit of a strange one, basically the library thinks that the payload type is different than it actually is. Should be quite harmless, but a bug regardless; I'll look into it.

If there's a way to help you debug, feel free to shout and I'll help where I can.

malmeloo, 2025-01-23:
I've been looking into this today but cannot reproduce the issue. Could one of you bump the log level in the `device_scanner.py` example to `DEBUG` and upload the logs it generates? It will likely generate a lot of data, so you may want to pipe it to a file using `python3 device_scanner.py &> logs.txt` (or whatever the windows equivalent is)

Do note that it contains the mac addresses of any bluetooth devices in your vicinity, which can *technically* be used to obtain a rough estimate of your location (see e.g. WiGLE). If you know what to look for feel free to remove any lines containing non-Apple mac addresses (ctrl+f for `{76:` to find which addresses belong to airtags), or email me at git {at} mikealmel.ooo if you prefer to share it privately.

thisiscam, 2025-01-29:
I haven't been able to reproduce this for some reason. Maybe it was some nearby device that's causing this but it's now gone...

malmeloo, 2025-01-30:
Hmm, well let me know if you run into it again. I'll keep trying every once in a while