malmeloo/FindMy.py #100: On-device code for the emitter

maelp, 2025-01-16:
Very nice code! Looks like there is everything to scan and monitor for FindMy, you could complement it with a Micropython implementation of the emitters for completeness? There is a working implementation at https://github.com/biemster/FindMy/blob/main/ESP32/findmy_apple.py (only thing is: for some reason only some keys are accepted by this implementation, and you need a modified micropython with a function to get the MAC address)

malmeloo, 2025-01-16:
Thanks! I think I'd rather keep this repository focused on fetching FindMy data, though. There are a lot of different firmwares, and I'm not sure how viable Micropython is for use in production, since I can imagine that its power usage would be much higher than C/C++ firmware. However, I do believe it would be interesting to document all the different firmwares that people have created and keep track of them somewhere. Right now development is very ad-hoc with no centralized place to find everything, other than Google.

malmeloo, 2025-01-16:
That's fair. It could be interesting to add a "payload generation" feature in the library somewhere which can be used to generate the BLE data to broadcast the tag. Or add another example that explains how to generate static keys and use them. Still lots to do in terms of documentation...

maelp, 2025-01-16:
Indeed, just having script that produce "broadcastable data" would be already good, and people could check the implementation that way

malmeloo, 2025-01-16:
Sounds good, I'll just reopen this issue then as a feature request.

commit 8f53b179 (2025-07-23): Add pubkey-ble conversion example

Resolves #100