malmeloo/FindMy.py #101: Key generation

maelp, 2025-01-16:
Your README mentions that the key generation is implemented, but in the source I only see an abstract class without implementation for key generation, am I missing something, or is it not yet implemented?

malmeloo, 2025-01-16:
Depends on what keys you want to generate.

Static key generation is done [here](https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/keys.py#L123) (it's just 28 random bytes)
Accessory subkey generation is implemented [here](https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/accessory.py#L130)

There is currently no way to generate new accessory master / skn / sks keys (to use as a "seed"), simply because there hasn't been a need to do that yet. I'm also not sure if there will ever be a need - key generation for standard-compliant accessories is done collaboratively, so there is no way for the iPhone to have full control over the selected master and secret keys. If you're looking into implementing your own key rollover, I'd recommend using a different algorithm that is more focused on your use case. Examples include simply using a static pool of keys, a KDF (much like what Apple is doing, see [here](https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/accessory.py#L241-L243)), or using some kind of hash chain with forward secrecy. If you implement it as a subclass of `RollingKeyPairSource`, you can simply use an instance of that class as an argument to the regular location-fetching function. That said, those three seed values are essentially just random bytes, so you could generate them manually if you really wanted to.

maelp, 2025-01-16:
Ah okay, I thought by quickly skimming other code that it was more complex to get a public / private / advertising key? here they do it with an elliptic curve, is it unnecessary? https://github.com/biemster/FindMy/blob/main/generate_keys.py#L26

maelp, 2025-01-16:
Ahhh sorry I just saw the code over here https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/keys.py#L98

closed 2025-01-16 by maelp

malmeloo, 2025-01-16:
My bad, I should've been more clear. Let me try to elaborate:

FindMy uses P-224 elliptic curve keys, yes. This library has already abstracted all of those elliptic curve details away, so you can simply generate a new "keypair" using `KeyPair.new()` and obtain its private/public key by calling one of its `{private,public}_key_{b64,bytes}` properties. Every device that uses FindMy uses such a key; its public key is transmitted to nearby devices. However, the difference is how the key is generated.

Most diy tags (OpenHaystack etc) use a single, static key that is transmitted at all times. This works and is super efficient when querying location reports, but it's not very privacy-friendly, since it makes you easier to track in real-life. For this reason, the official spec (which all Apple and 3rd-party FindMy devices adhere to) uses an algorithm that is effectively seeded by one master key and two secret keys (sks/skn). Depending on the state of the device (nearby, separated, etc.) it will then broadcast a public key that is generated from those seeds and the current time, making it effectively change its identity to any outside observer. The main thing to realize here is that while outsiders cannot predict the device's next identity, the owner can, allowing them to query the device's location by submitting a bunch of those identities to Apple.

So essentially, for basic personal use, simply using a static key is fine. If you want to deploy it on multiple devices, then you should probably use a key rollover strategy like Apple does. However, their algorithm isn't the most efficient (in my opinion), and you can achieve equivalent forward secrecy using different algorithms that will be much more readable.

maelp, 2025-01-16:
Okay very clear! Do you have some ideas / pointers to those other forward secrecy algorithms?

Also I'm not very clear on how the device and the owner can "synchronize" their keys in absence (as is often the case) of a precise clock on the lost device? Does the "finder" location report come with a rough timestamp and the owner just generates a bunch of keys that might correspond to those around those timestamps?

malmeloo, 2025-01-16:
Apple essentially uses two of those algorithms in parallel, as can be seen in the link I posted above ([this one](https://github.com/malmeloo/FindMy.py/blob/5cd53fbed9f9d676a4fda855bfa779df84dd3d28/findmy/accessory.py#L241-L243)). They're essentially just taking the random seed and applying a KDF a bunch of times with their constant `"update"` string. You could totally just copy that part of the design. Also, I'm not a cryptography expert, but I think using a single hash function and seed would be sufficient as well. If you take `s_0` to be the initial seed and define `s_i = SHA256( s_(i-1) )`, then output `k_i = SHA256( s_0 xor s_i )` as secret key, it would probably satisfy forward secrecy as well. But if you want to be 100% sure without having to trust a stranger on the internet, look into using KDFs instead: https://en.wikipedia.org/wiki/Key_derivation_function.

malmeloo, 2025-01-16:
Oh and about key synchronization: owner devices synchronize the system time with their accessories. The time at which they're paired is `i = 0`, and in every "time slot" the index is incremented. That way both the accessory and device will derive the same key, since they agree on the timeslot.

maelp, 2025-01-16:
Ah I see! So each report has an associated timestamp?

malmeloo, 2025-01-16:
Each downloaded report has two timestamps: one of when the tag was found, and one indicating when the report was uploaded. But the tags themselves don't actually broadcast a timestamp; all of that is handled by Apple and their devices. The tags just broadcast that public key and a few bytes of additional data.

maelp, 2025-01-17:
Small question (sorry for asking it here, but couldn't find the info in a quick Google search), do you know what the "confidence" is? since it's not related to horizontal_accuracy, I'm not clear on what it means for FindMy? Is it because there might be collisions in the report?

malmeloo, 2025-01-18:
I'm really not sure, and the original OpenHaystack paper doesn't appear to mention it either. If you ever find out then please let me know 🙂