malmeloo/FindMy.py #102: com.apple.mobileme login failed with status 1: Account limit reached.

maelp, 2025-01-17:
Trying to log in with an account and I'm getting this. I'm wondering if it is because of the headers describing the machine I'm using to login? We're supposed to be able to have at least 3 accounts per device no? And in my case I'm only using one other. Any idea how to solve this?

malmeloo, 2025-01-18:
Interesting, I've never seen this error before. Can you try using a different Anisette server? My guess is that there is a limit to the number of accounts you can attach to a device, and Apple probably tracks this using the machine's identifier which is generated by Anisette.

maelp, 2025-01-18:
So you mean deleting the volume and recreating it? I'll try

fz6, 2025-01-19:
I am experiencing the same issue. I am using an Apple ID I created many years ago, that had a real iPhone linked to it (so just 1 device). I am seeing new devices be added, but receiving the same error:

```
INFO:root:Transitioning login state: LoginState.REQUIRE_2FA -> LoginState.AUTHENTICATED
INFO:root:Logging into com.apple.mobileme
Traceback (most recent call last):
  File "/home/mike/Desktop/AirTag/FindMy.py/examples/_login.py", line 79, in get_account_sync
    with acc_store.open() as f:
  File "/usr/lib/python3.10/pathlib.py", line 1119, in open
    return self._accessor.open(self, mode, buffering, encoding, errors,
FileNotFoundError: [Errno 2] No such file or directory: 'account.json'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mike/Desktop/AirTag/FindMy.py/examples/fetch_reports.py", line 38, in <module>
    sys.exit(fetch_reports(sys.argv[1]))
  File "/home/mike/Desktop/AirTag/FindMy.py/examples/fetch_reports.py", line 17, in fetch_reports
    acc = get_account_sync(
  File "/home/mike/Desktop/AirTag/FindMy.py/examples/_login.py", line 82, in get_account_sync
    _login_sync(acc)
  File "/home/mike/Desktop/AirTag/FindMy.py/examples/_login.py", line 42, in _login_sync
    method.submit(code)
  File "/home/mike/Desktop/AirTag/venv/lib/python3.10/site-packages/findmy/reports/twofactor.py", line 211, in submit
    return self.account.sms_2fa_submit(self._phone_number_id, code)
  File "/home/mike/Desktop/AirTag/venv/lib/python3.10/site-packages/findmy/reports/account.py", line 1006, in sms_2fa_submit
    return self._evt_loop.run_until_complete(coro)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/mike/Desktop/AirTag/venv/lib/python3.10/site-packages/findmy/reports/account.py", line 537, in sms_2fa_submit
    return await self._login_mobileme()
  File "/home/mike/Desktop/AirTag/venv/lib/python3.10/site-packages/findmy/reports/account.py", line 825, in _login_mobileme
    raise UnhandledProtocolError(msg)
findmy.errors.UnhandledProtocolError: com.apple.mobileme login failed with status 1: Account limit reached.

```

I had only setup the Anisette server today, but I tried:

```
sudo docker stop anisette-v3
sudo docker rm anisette-v3
sudo docker rmi dadoum/anisette-v3-server:latest
sudo docker run -d --restart always --name anisette-v3 -p 6969:6969 dadoum/anisette-v3-server:latest
```

And still receive the same error

fz6, 2025-01-19:
I found a thread related to this elsewhere: https://github.com/Chapoly1305/FindMy/issues/9

My girlfriend had an old iPad sat in a cupboard so I charged it up and logged in - now my Apple ID works 👍

maelp, 2025-01-19:
I tried deleting the anisette volume and still had the issue

maelp, 2025-01-19:
@fz6 so you mean it's because you need a recent login?

maelp, 2025-01-19:
Indeed I created a new user on my mac and logged in with the burner account, and then I could log in on anisette

closed 2025-01-19 by maelp