malmeloo/FindMy.py #107: Installation

cyber-vi-king, 2025-02-02:
Hi, I run Home Assistant 2025.1.4 in docker and don’t have HACS. 

I tried to install this app by placing the findmy folder in the custom_components folder into my config/custom_components folder. But after a restart of Home Assistant I don’t see anything under integrations nor any obvious logs in the gui. 

What am I missing?

Best regards

malmeloo, 2025-02-02:
Installation methods other than the listed method (through HACS) are not supported, and might not work correctly. I'd recommend simply installing HACS and following the official instructions. I don't have the time to help debug all kinds of ad-hoc installation methods, especially while the integration is still very experimental, sorry. But if you have a concrete issue with debug logs feel free to create an issue (in the correct repository 😉) and I'll take a look.

closed 2025-02-02 by malmeloo

cyber-vi-king, 2025-02-04:
I just noticed I managed to make an issue in the wrong project altogether. For your information, just copying the findmy folder in the HA custom_components folder is enough (HACS doesn't do much more I guess) - the integration doesn't show up though since it needs to be configured. So I was missing the step "click add integration and find it in the list in order to get to the configuration dialog". Don't know if that's intentional, since you state it's still in early development.

Just wanted to add this information for you. I managed to set it up without HACS. My newly created Apple account threw some weird errors, but since my "daily driver" account works, I guess there is still something hacky with my test account and not anisette or your project.

malmeloo, 2025-02-04:
Ahh clear, I thought you meant that you couldn't find the integration in the list to add. Every integration needs to be added that way first, since HA will only show an integration in the overview if it has at least one instance. Glad you got it to work though!

As for the new account issue, you need to sign into an Apple device at least once before it can be used. After that you should be able to use the new account as normal.