malmeloo/FindMy.py #108: Can this be used with compatible tags?

cqcopilot1, 2025-02-03:
Hi! First of all, thanks a lot for the project and your hard work with it 👏 

I wanted to ask about whether this would fit my purpose. I have a Mac, but I use Android.
I bought a couple cheap Find my compatible trackers for 5$ on AliExpress, and I thought I would be able to use this project to set them up.

I have looked into this and macless-haystack, but I'm stuck at the part of enrolling it.
As this is a pre-made tag, I cannot flash it. Or should I tear it down and flash it? Did I completely misunderstand the project?

Thanks!

malmeloo, 2025-02-03:
This project can be used to query locations for already-enrolled tags by dumping their info from the FindMy app on a Mac. Unfortunately it cannot be used to set up new airtags, as this process requires communication with Apple servers, which has not been reverse engineered yet. It's also not trivial to bypass, since the accessory verifies that this communication did indeed occur using public key cryptography.

I would recommend using one of the many OpenHaystack options to create a custom tag if you have no way to set up a real/compatible tag. You can also ask a friend or family member to do it and sign into their account on your Mac to obtain the required plist file, then disconnect the tag from their account (though I'm not sure if this resets the airtag). But do note that this is currently not possible to do on macOS 15 (#31) and there appear to be some issues with airtag support in general (#90). So while flashing your own tag may take more time to do, it can also prevent quite a few headaches.

malmeloo, 2025-02-15:
Closing this issue because I think your question should be answered, if not feel free to reopen.

closed 2025-02-15 by malmeloo