malmeloo/FindMy.py #111: Usage and installation

beckmx, 2025-02-13:
Hello there, I was checking your docs, I am using a mac and installed the pip library, if i understand correctly I install the library through pip and then use it as in the examples, correct? but i also noticed the anisette dependency inside the examples, just wondering if you have a reference on installing anisette correctly

greetings!

malmeloo, 2025-02-13:
Anisette refers to an Anisette provider such as [anisette-v3-server](https://github.com/Dadoum/anisette-v3-server). It's a server you can host on your own computer and need to point FindMy.py towards.

In the near future this requirement will be lifted and the library will be able to obtain Anisette data itself, which should make installation a lot easier. However, currently this is not possible yet.

malmeloo, 2025-03-01:
Those errors can be ignored, they just mean the underlying sessions haven't been closed properly. If it doesn't print anything then it didn't find any reports, so is your tag still working? Apple only stores reports for up to 7 days.

closed 2025-06-14 by malmeloo