malmeloo/FindMy.py #120: Question regarding importing open haystack devices

aaronjrodrigues, 2025-03-21:
Quick question, I'm planning on purchasing a microbic to be able to flash open haystack on. I wanted to ask if it is possible to be able to import that open haystack device into the find my app itself so that I can view it on the find my app on my devices. In addition to that, if that is not possible. Does FindMy.py also have any way of importing these devices into home assistant?

Any help would be appreciated, thanks.

malmeloo, 2025-03-21:
I don't use and have never used OpenHaystack, so I wouldn't know the answer to your first question. However, my hass-FindMy integration (see README) supports static-key tags in its current state. So you can simply enter the private key you used to deploy the device, and the integration will create device tracker entities for you.

aaronjrodrigues, 2025-03-21:
> I don't use and have never used OpenHaystack, so I wouldn't know the answer to your first question. However, my hass-FindMy integration (see README) supports static-key tags in its current state. So [...]

Just to clarify, when you say 'create tracker entities' does it mean it'll create it into the find my app of that device or is it in the script itself?

malmeloo, 2025-03-21:
It will create them in Home Assistant. So Home Assistant will give you device tracker entities to show on a map, do automations with, etc. It's similar to the entities tracking your other devices (phones, etc.), except it tracks your custom FindMy tags instead.

aaronjrodrigues, 2025-03-25:
@malmeloo quickly kqeustion I got find my for homeassistant setup however im experiencing a weird issue where it'll import my open haystack device and show the location however after a few minutes 10-15 minutes. The status of my open haystack tracker will go to 'Unavailable' in home assistant.

Please find the attached screenshot:

[image]

Please also find the attached debug logs:

[home-assistant_findmy_2025-03-25T13-35-33.864Z.log](https://github.com/user-attachments/files/19448842/home-assistant_findmy_2025-03-25T13-35-33.864Z.log)
[home-assistant_findmy_2025-03-25T10-23-18.262Z.log](https://github.com/user-attachments/files/19448843/home-assistant_findmy_2025-03-25T10-23-18.262Z.log)

Any help would be appreciated, I'm really loving the integration - it's just this one issue. Thanks.

malmeloo, 2025-03-25:
The logs seem to indicate that the integration is losing its login session. This is usually caused by your anisette configuration.

Are you using your own anisette server? Public anisette servers are not supported. Also make sure that the anisette server you are using retains its configuration, as wiping it will cause the exact same issue as you are facing now.

aaronjrodrigues, 2025-03-25:
> The logs seem to indicate that the integration is losing its login session. This is usually caused by your anisette configuration.
>
> Are you using your own anisette server? Public anisette servers [...]

ok I tried creating my custom server but it seems that the issue happens again. https://docs.sidestore.io/docs/advanced/anisette/ I followed this guide to create a custom server.

malmeloo, 2025-03-25:
Could you try using https://github.com/Dadoum/anisette-v3-server instead? I haven't tested the sidestore server myself, so I'm not sure how it behaves. It might wipe its config routinely since it's intended to be used as a public server.

aaronjrodrigues, 2025-03-25:
> Could you try using https://github.com/Dadoum/anisette-v3-server instead? I haven't tested the sidestore server myself, so I'm not sure how it behaves. It might wipe its config routinely since it's intended to be used as a public server.

The docs that I followed told me to install the same one (https://github.com/Dadoum/anisette-v3-server) and I'm having issues on that. 

Here's some more debug logs in case it has some extra information, thank you again for your support:

[home-assistant_findmy_2025-03-25T22-46-19.709Z.log](https://github.com/user-attachments/files/19457845/home-assistant_findmy_2025-03-25T22-46-19.709Z.log)

malmeloo, 2025-03-25:
Ah whoops, I missed that part. Assumed it was a custom one. It's pretty late here right now, so I'll take another look tomorrow if you don't mind.

aaronjrodrigues, 2025-03-25:
> Ah whoops, I missed that part. Assumed it was a custom one. It's pretty late here right now, so I'll take another look tomorrow if you don't mind.

No problem mate, take your time.

Cheers,

malmeloo, 2025-03-26:
Your log seems to be absolutely cluttered with errors from this integration, but one of them caught my eye. The rest are most likely caused by this error and raised due to bad error handling from me.

`findmy.errors.UnhandledProtocolError: com.apple.mobileme login failed with status 1: Account limit reached.`

This most likely means that the Apple ID you are using has never been used on a real Apple device. That's unfortunately required for this integration, since Apple only enables the `mobileme` API (which this integration uses) for your account after you have used it on a device at least once. *Usually* they show a different error in this case, but they seem to like switching up the error messages every once in a while.

There's a little more info about this issue here: https://github.com/malmeloo/FindMy.py/issues/102

malmeloo, 2025-03-26:
Oh also, I can see your email in at least one of the logfiles. Not sure to what extent you want to keep that secret, but you might want to remove them.

aaronjrodrigues, 2025-03-26:
thats quite interesting as I did try multiple accounts - which didn't work because they didn't have an Apple device linked to them.

However, the main account that I have (around 4 devices linked) is what gives the previous behaviour where it works for a few hours then it doesn't work.

-aaron

aaronjrodrigues, 2025-03-26:
whoops my bad thanks

> Oh also, I can see your email in at least one of the logfiles. Not sure to what extent you want to keep that secret, but you might want to remove them.

aaronjrodrigues, 2025-03-26:
> Oh also, I can see your email in at least one of the logfiles. Not sure to what extent you want to keep that secret, but you might want to remove them.

Hi I've attached a separarate log file which I think might have some extra info as I enabled debug logging before logging into my account. I've also redacted sensitive info. Thanks.

[home-assistant_findmy_2025-03-26T01-12-13.819Z.log](https://github.com/user-attachments/files/19475751/home-assistant_findmy_2025-03-26T01-12-13.819Z.log)

malmeloo, 2025-03-26:
Hmm so I just remembered an issue similar to yours: https://github.com/malmeloo/FindMy.py/issues/48#issuecomment-2223811896

Could you try changing your Anisette setup by specifying the volume as indicated above? You will probably have to remove and re-add the account in Home Assistant after doing so. I'm still leaning towards an Anisette issue here.

aaronjrodrigues, 2025-03-26:
Okay I've tried changing it, fingers crossed that is works. I'll keep you posted.

Thanks.

aaronjrodrigues, 2025-03-28:
> Hmm so I just remembered an issue similar to yours: [#48 (comment)](https://github.com/malmeloo/FindMy.py/issues/48#issuecomment-2223811896)
> 
> Could you try changing your Anisette setup by specifying the volume as indicated above? You will probably have to remove and re-add the account in Home Assistant after doing so. I'm still leaning towards an Anisette issue here.

hi, unfortunately it seems the issue still happens :(

malmeloo, 2025-03-28:
Ah dang, I was really hoping that'd fix it. Unfortunately I don't really know what else could be wrong... how many accounts do you have set up? a single one or multiple? If you're using multiple accounts on a single anisette server you could try using multiple servers instead. But otherwise I think this might be a case of "Apple doesn't like/trust your account."

aaronjrodrigues, 2025-03-30:
> Ah dang, I was really hoping that'd fix it. Unfortunately I don't really know what else could be wrong... how many accounts do you have set up? a single one or multiple? If you're using multiple [...]

I have one account, but quick question - the way I have it setup on open haystack is using my search party key. is there any way to import my search party key into find my.py? thanks again for all your help.

malmeloo, 2025-03-30:
It's not officially supported, but you might be able to replace the token in the json returned by `account.export()` and then re-import that data using `account.restore()`. I think that should work until the token expires.

I've received another report from someone with the same issue as you, so unfortunately this appears to be a recurring problem. I'll try to look into it when I find the time.

aaronjrodrigues, 2025-04-06:
> It's not officially supported, but you might be able to replace the token in the json returned by `account.export()` and then re-import that data using `account.restore()`. I think that should work [...]

Thank you so much