malmeloo/FindMy.py #143: Unable to fetch original tag data using fetch_reports

gokhanozturk, 2025-07-07:
Previously, I was able to fetch original tag data using fetch_reports after loading the private key extracted via the Genom code. However, it now returns only empty data. Am I missing something or doing something wrong?

```
python3 fetch_reports.py <redacted_private_key>
Logged in as: <redacted_email> (<account_name>)
{KeyPair(name="None", public_key="<redacted_public_key>", type=KeyType.UNKNOWN): KeyPair(name="None", public_key="<redacted_public_key>", type=KeyType.UNKNOWN)}
INFO:findmy.reports.anisette:Fetching anisette data from http://localhost:6969
defaultdict(<class 'list'>, {})
```

malmeloo, 2025-07-08:
Also, what do you mean with "Genom code" exactly? If you are using an original airtag you should not be using that script, but the `real_airtag.py` script instead.

gokhanozturk, 2025-07-08:
Thanks for the explanation. I'll try your suggested method and see how it goes.

malmeloo, 2025-07-10:
Okay, well that key should have never even worked with the `fetch_reports.py` script in the first place. That one is only for tags that never rotate their keys, which official and third-party airtags do. So unless the key you found is one of the airtag's rotating keys, I'm not sure why it worked for you before.

Which version is giving you the `privateKey` error? Can you try the `real_airtag.py` script with both the one in that PR and v0.7.6 (from releases/pypi, so not the latest version)?

closed 2025-08-25 by malmeloo