malmeloo/FindMy.py #154: Decrypt the power information of the device from the obtained encrypted data

runfengXie, 2025-07-22:
@malmeloo Hi!

Can the current battery informationof the device be decrypted from the device positioning encryption data returned so far?

I want to be able to get the device's power information when the device is far away, so that I can know in time whether the device is going to be powered off, and I can change the power in time. Or is there any other better way to solve how to determine whether the equipment is powered off? If you only use positioning data, you may misjudge it, because there may be no equipment around you to report positioning.

runfengXie, 2025-07-22:
[image]
This is the response result, just before can payload be decrypted?

runfengXie, 2025-07-22:
I started by trying to define it myself. I took a look at your firmware implementation and data decryption code. Thank you very much for your answer!

malmeloo, 2025-07-22:
I don't have a firmware implementation unfortunately, but you can find the decryption logic here: https://github.com/malmeloo/FindMy.py/blob/main/findmy/reports/reports.py

Good luck!

runfengXie, 2025-07-22:
Thank you so much!

malmeloo, 2025-08-04:
I think this question should be answered, feel free to reopen if not

closed 2025-08-04 by malmeloo