malmeloo/FindMy.py #161: Will concurrent requests cause them to be discovered by Apple servers?

runfengXie, 2025-08-14:
I wonder if multiple requests for different keys are sent at one time will be blocked by Apple's server. I dare not try at present, and I want to ask if there is this risk.

parawanderer, 2025-08-15:
I look up official AirTags [in a one-by-one loop](https://github.com/parawanderer/OpenTagViewer/blob/47f7fe63cd1b1a14531eca0378604f07cb243dfd/app/src/main/python/main.py#L211) [in my app](https://github.com/parawanderer/OpenTagViewer) every minute that my app is actively being viewed and have yet to get banned. Well, I don't have that many AirTags in the first place at 4.

malmeloo, 2025-08-16:
It's recommended to pass multiple tags as a list to `fetch_reports`, that way the library can optimize fetches to use as few requests as possible. However the official MacOS app also just sequentially polls tags, so realistically either should be fine. I wouldn't do the requests concurrently though.

runfengXie, 2025-08-16:
I have tried multiple lists, but the reports returned are missing; I don't know if it's just a single request, that's why I'm considering whether it's multiple requests at the same time and whether they'll be detected by the authorities.

Are you not recommending concurrency now.

malmeloo, 2025-08-16:
Yeah that's an issue with the current pypi version. You can either switch to the main branch of this repository or just make the calls sequentially. I just wouldn't make them concurrently; be nice to apple's servers (and the library is not guaranteed to be thread safe anyway)

runfengXie, 2025-08-16:
OK. Thank you for your answer!

closed 2025-08-25 by malmeloo