malmeloo/FindMy.py #176: examples/real_airtag.py is asking for an airtag_path?

victorhooi, 2025-09-16:
I'm trying to run the example scripts (e.g. `examples/real_airtag.py`), and it's giving me an error message asking for the airtag path - however, it's a little unclear what this airtag path is? (Struggling a bit with the documentation here, and trying to figure it all out from Github issues...lol).
```
> uv run examples/real_airtag.py
/Users/victorhooi/code/FindMy.py/.venv/lib/python3.9/site-packages/urllib3/__init__.py:35: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
/Users/victorhooi/code/FindMy.py/.venv/lib/python3.9/site-packages/fs/__init__.py:4: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  __import__("pkg_resources").declare_namespace(__name__)  # type: ignore
usage: real_airtag.py [-h] airtag_path
real_airtag.py: error: the following arguments are required: airtag_path
```

I did see mention in [this issue](https://github.com/malmeloo/FindMy.py/issues/31), about using the Swift program [here](https://gist.github.com/airy10/5205dc851fbd0715fcd7a5cdde25e7c8) or [here](https://gist.github.com/YeapGuy/f473de53c2a4e8978bc63217359ca1e4) to decrypt the plist file - however, I should mention that Swift script also didn't work for me:

```
Swift/ErrorType.swift:254: Fatal error: Error raised at top level: main.MyError.noPassword
fish: Job 1, './airtag-decryptor' terminated by signal SIGTRAP (Trace or breakpoint trap)
```

I don't know if that's expected due to recent changes in macOS?

I do have access to a real macOS device.

I've cloned the repo, and done a `uv sync`. (I had to install `cmake` and `pkg-config` to get it to build correctly - I assume this is related to `unicorn`?)

Is there an up-to-date list of the steps (from scratch) of how to run the examples (like `real_airtag.py`), on either a macOS device, or a non macOS/Apple device?

(I assume the the script is trying to call the `from_json` [here](https://github.com/malmeloo/FindMy.py/blob/b18d4b5c0c9d5d98ede33e5a6363581ea8f76e35/findmy/accessory.py#L340) against a path it expects - but it's unclear what paths I should be passing in here).

malmeloo, 2025-09-16:
What MacOS version + architecture are you on? You shouldn't need to manually build unicorn if a wheel for it exists, but I guess uv/pip is not picking up on it.

I agree that dumping the airtags is quite bothersome and confusing, even for me. I only have a Sonoma VM to test with, and Apple keeps changing how airtags are stored/encrypted. There are plans to pull them directly from Apple servers (#173), but this will take quite a lot of work to implement.

Can you try using the library's builtin decryptor? I'm not familiar enough with the swift code to know what the issue is you're running into. -> `python3 -m findmy decrypt`

victorhooi, 2025-09-16:
I'm testing this on macOS 26.0 (25A353) (i.e. Tahoe), on Apple Silicon.

If you try to run `uv sync` from the repo on a fresh macOS machine, you get the error complaining about `cmake`:

https://gist.github.com/victorhooi/bea6ea752d888e1ae9b35fff034b772a

Then, after installing `cmake` - it also complains about `pkg-config`:

https://gist.github.com/victorhooi/f6889e6a39ca6535a032a337e4edc8cb

Anyway, I installed both `cmake` and `pkg-config` on the box, and it was able to build unicorn. 

This is what I get if I try to run `findmy decrypt`:
```
victorhooi@mbp16-personal ~/c/F/findmy (main)> uv run python3 -m findmy decrypt
/Users/victorhooi/code/FindMy.py/.venv/lib/python3.13/site-packages/fs/__init__.py:4: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  __import__("pkg_resources").declare_namespace(__name__)  # type: ignore
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/victorhooi/code/FindMy.py/findmy/__main__.py", line 105, in <module>
    main()
    ~~~~^^
  File "/Users/victorhooi/code/FindMy.py/findmy/__main__.py", line 79, in main
    decrypt_all(args.out_dir)
    ~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/Users/victorhooi/code/FindMy.py/findmy/__main__.py", line 99, in decrypt_all
    accs = list_accessories()
  File "/Users/victorhooi/code/FindMy.py/findmy/plist.py", line 120, in list_accessories
    key = _get_beaconstore_key()
  File "/Users/victorhooi/code/FindMy.py/findmy/plist.py", line 37, in _get_beaconstore_key
    return bytes.fromhex(key_in_hex)
           ~~~~~~~~~~~~~^^^^^^^^^^^^
ValueError: non-hexadecimal number found in fromhex() arg at position 0
```
Should I disable SIP before trying your command?

victorhooi, 2025-09-17:
I did try with beaconstore-extractor - you can see my results in [this issue](https://github.com/pajowu/beaconstorekey-extractor/issues/5).

(I did disable SIP for this part - as confirmed with `csrutil status` - both for trying with beaconstoreKey-extractor, and also with running your `findmy decrypt`).

If I look in Keychain Access, I did see an item named "BeaconStore" if that helps at all:

[image]

victorhooi, 2025-09-17:
I can't seem to get it to reveal that item.

I do have a "Beaconstore" item in both the `login` keychain, and also the `iCloud` keychain:

[image]

If I click on "Show password", I am prompted for my macOS login password - but even if I enter that in, and hit OK - nothing shows in the password text box:

[image]

I do note that on the second page, under Access Control, it lists `com.apple.icloud.searchpartyuseragent` as the only access group with permission:

[image]

malmeloo, 2025-09-17:
I tried `uv sync`ing the project in my MacOS VM and it installs unicorn from the wheel without building. [It looks like](https://pypi.org/project/unicorn/#files) unicorn should have wheels for apple silicon as well, so I'm not sure why it decided to build from source for you. I think I'll just release it as is, we'll find out soon enough if other people are also having installation issues.

I've made a pinned issue to track macos (in)compatibility in #177, and I almost have documentation ready that will explain how to dump accessories for different versions of MacOS. I'd propose to share future findings there to keep things central.

closed 2025-09-17 by malmeloo